nkolban/esp32-snippets #916: Trying to loop through identical bluetooth clients, hangs?

coogle, 2019-10-16:
I'm trying to write an Arduino client for a BLE-enabled battery that basically loops through each battery it can find in a scan, gets the data from it, and repeats (to be published via MQTT).. What I'm finding is that this works once but there is this weird issue where sometimes calling `BLEClient::connect()` causes an immediate disconnect and then the app just hangs without any further output. 

You can see my code here (check BatteryManager.cpp for most of it):

https://github.com/ThisSmartHouse/LiFeBlue-BLE

This is the output I get when I run the code in full (it just hangs and does nothing past this point):

```
LiFeBlue ESP32 Battery Monitor
(c) 2019 Internet Technology Solutions / This Smart House
http://www.thissmarthouse.com/lifeblue
-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-

- Initialized BLE Client
- Starting BLE Device Scan...
[V][BLEScan.cpp:195] start(): >> start(duration=10)
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: ScanEnd (0x3ffdde24), owner: <N/A> for start
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: ScanEnd (0x3ffdde24), owner: start
[V][BLEScan.cpp:227] start(): << start()
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 2 0x02
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 7 0x07
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -61
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 05
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x2
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x02 (), length: 2, data: e0ff
[D][BLEAdvertisedDevice.cpp:447] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000ffe0-0000-1000-8000-00805f9b34fb
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 8, data: 00007c010a41d552
[D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 00007c010a41d552
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x9
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 14, data: 4c4231323130302d50432d353730
[D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: LB12100-PC-570
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -87
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 10, data: 4c0009060207c0a801bd
[D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 4c0009060207c0a801bd
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -74
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 10, data: 4c0009060306c0a8011a
[D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 4c0009060306c0a8011a
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -80
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 6, data: 4c0010020b00
[D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 4c0010020b00
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[... 1166 lines cut ...]
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
Client for 7c:01:0a:41:d5:52 disconnected
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: OpenEvt (0x3ffdecd4), owner: <N/A>
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: RssiCmplEvt (0x3ffde1c8), owner: <N/A>
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: SearchCmplEvt (0x3ffddf0c), owner: <N/A>
[I][BLEDevice.cpp:604] removePeerDevice(): remove: 0, GATT role client
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
```

What I expect it to be doing is:

- Adding all the batteries it knows about to the processing queue
- Connect to the battery
- Register a notification callback on a charactistic
- Start getting notifications (which consist of chunks of data)
- Store the chunks of data in a CircularBuffer until we capture it all
- Once we capture a full data block, disconnect from the server
- Once we are in a state where we are not currently connected to a battery, pop the next battery off the processing stack and start again

chegewara, 2019-10-16:
Hi,
did you use most recent arduino library? There is fix preventing semaphore stuck.

coogle, 2019-10-16:
Unless I am supposed to use something different than what is provided by Aruduino IDE. I am using the ESP32 Arduino library v 1.0.4 which claims to be the latest.

chegewara, 2019-10-16:
Ok, please make sure this patch is included in library:
https://github.com/espressif/arduino-esp32/pull/2974

Sometimes library is not updated when you update arduino-esp32.

coogle, 2019-10-16:
Yep it seems included in the library as far as I can see. Although in my version BLERemoteCharacteristic.cpp:403 returns `std::string()` instead of `NULL` -- i assume that was patch again later.

Either way, this is what I have for BLEClient.cpp that's relevant (around line 184) and looks like it includes the patch:

```
                case ESP_GATTC_DISCONNECT_EVT: {
                                // If we receive a disconnect event, set the class flag that indicates that we are
                                // no longer connected.
                                m_isConnected = false;
                                if (m_pClientCallbacks != nullptr) {
                                        m_pClientCallbacks->onDisconnect(this);
                                }
                                esp_ble_gattc_app_unregister(m_gattc_if);
                                m_semaphoreOpenEvt.give(ESP_GATT_IF_NONE);
                                m_semaphoreRssiCmplEvt.give();
                                m_semaphoreSearchCmplEvt.give(1);
                                BLEDevice::removePeerDevice(m_appId, true);
                                break;
                } // ESP_GATTC_DISCONNECT_EVT
```

Is there anything I can do to help debug this? I would be happy to contribute more than complaints if I can find a way to help!

coogle, 2019-10-16:
Also FWIW It seems that there is a decent lag right now between the "official" ESP32 library and this one. I noticed that in yours it seems that `getRemoteService()` in `BLERemoteCharacteristic` has been moved to `public` while the version of the library I am using has it annoying marked as `private` (which made it difficult to know which device was triggering my notification callback.

chegewara, 2019-10-16:
I have to think about it. Seems that some semaphore still causing code stuck. But im not working with this library now, so i cant say how long it can take. Sorry.