nkolban/esp32-snippets #918: Connecting to a BLE peripheral with previously saved credentials (BLEAddress, esp_ble_addr_type_t)

axa88, 2019-10-23:
Might someone be able to help me understand the Bluetooth connect prerequisites.

What i am do now successfully is use the esp32 as a BLE central instantiating a `BLEClient` which scans for BLE advertisements, and when identifying the correct device, saves an instance of that `BLEAdvertisedDevice` to be used as the parameter to the Client's Connect method.

`connect(BLEAdvertisedDevice* device);`

From this BLEAdvertisedDevice object I can get the 'BLEAddress'
And when that device connects, i can gain the `esp_ble_addr_type_t`.

So shouldn't it then be possible to save these 2 parameters in non volatile memory and use them after a program reset, or without ever additionally scanning for this device using the other connection overload?

`connect(BLEAddress address, esp_ble_addr_type_t type = BLE_ADDR_TYPE_PUBLIC);`

Sort of a direct connect, but at least having once connected to gain the 2 parameters.
but I fear i may not understand correctly because i thought advertised addresses could change otherwise. 

Ive not been able to do this yet, Ive read this old thread response by @chegewara stating that there may not be a way to actually direct connect without scan, but perhaps i misunderstand:
[](https://www.esp32.com/viewtopic.php?t=6965)

please advise

thomastech, 2019-10-24:
+1. I'm interesting in doing this on my client to connect to a BLE server.  Hopefully there is a way to do it.

My BLE client is working using the scan method. But I'd like to avoid performing a new scan each time it loses connection. Especially since scan sometimes experiences CPU hangs. Plus each new connect creates 200+ byte memory leaks.

- Thomas

axa88, 2019-10-24:
> @axa88 it should work without need to scan each time, but it will work only with peer device with public address. Just like you said, with private address it may change between connections and wont work.

So unfortunately I find when the remote device (android phone) connects the `BLEAdvertisedDevice` reports 
addr type: 1

`esp_ble_addr_type_t` 
`BLE_ADDR_TYPE_RANDOM`

@chegewara is this what you meant by "private address?
From your experience, with android devices is there any way to keep it from changing address, and thus be connectable even though address is random type?

closed 2019-10-26 by axa88

chegewara, 2019-10-26:
Yes, if you want to connect to android phone then it is using private address for sure.

Maybe, just maybe, because i didnt try it, you can use old device address when its paired/bonded device.

zachtos, 2020-07-02:
How about this procedure but for an ESP32 connecting to another ESP32?  I am trying the same thing, where I press a button to pair, and then it will always autoconnect to that device, because it maybe in a room with many other BLE server that are identical otherwise and I want to pair to the same one each time, but able to re-pair later w/o hard code change.