nkolban/esp32-snippets #919: ESP 32 readValue() function affecting IMU sensor data

tama-ts, 2019-10-25:
Hello!

I have set up a BLE ESP32 server and a BLE ESP32 client on two different ESP devices. Both of my devices collect data from IMU (Troyka) Sensors. The data collected on the server ESP (from the IMU) is transferred to the client ESP where the client reads that data, and does some calculations based on the readings from its own IMU. Now I have a problem where the data collected from the sensor which is connected to the client ESP is wrong.

Further I have done a lot of debugging of the code and hardware and know the following:
1. The hardware works (I tested it with multiple devices and different code)
2. The client receives the IMU data from the server ESP really well and quickly - and then further transfers it to my computer through bluetooth really well. 
3. The client is able to read and transfer correct IMU data without one particular line that I will describe below.
3. I am 99.99% certain that the line in my client which is alltering my IMU data is:
std::string value = pRemoteCharacteristic->readValue();
Because when I comment it out my client reads the IMU data correctly
4. I believe somewhere in the readValue function there is some code which affects my code for my IMU readings and I am not sure where. 

I will attach all of my code, thank you very much in advance! 

`#include "BluetoothSerial.h"
 
BluetoothSerial SerialBT;
// библиотека для работы с модулями IMU
#include <TroykaIMU.h>
 
// множитель фильтра
#define BETA 0.22
 
// создаём объект для фильтра Madgwick
Madgwick filter;
 
// создаём объект для работы с акселерометром
Accelerometer accel(ACCEL_ADDRESS_V1);
// создаём объект для работы с гироскопом
Gyroscope gyro(GYRO_ADDRESS_V1);
// создаём объект для работы с компасом
Compass compass(COMPASS_ADDRESS_V1);

// если напаяны перемычки, устройства доступны по новым адресам
// Accelerometer accel(ACCEL_ADDRESS_V2);
// Gyroscope gyro(GYRO_ADDRESS_V2);
// Compass compass(COMPASS_ADDRESS_V2);

// переменные для данных с гироскопа, акселерометра и компаса
float gx, gy, gz, ax, ay, az, mx, my, mz;
 
// получаемые углы ориентации (Эйлера)
int yaw, pitch, roll;
 
// переменная для хранения частоты выборок фильтра
float fps = 100;
 
// калибровочные значения компаса
// полученные в калибровочной матрице из примера «compassCalibrateMatrixx»

/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * author unknown
 * updated by chegewara
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("beb5483e-36e1-4688-b7f5-ea07361b26a8");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
[... 172 lines cut ...]
    }
    doConnect = false;
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected) {
    //String newValue = "Time since boot: " + String(millis()/1000);
    std::string value = pRemoteCharacteristic->readValue();
    //Serial.println(value.c_str());
    Serial.println(txString);
    //int i = atoi(value.c_str()-pitch);
    //SerialBT.println(i);
    //SerialBT.print("Here is recieved value: "); SerialBT.println(value.c_str());
    SerialBT.print("Here is the pithc we have: "); SerialBT.println(pitch);
    //Serial.println("Setting new characteristic value to");
    
    // Set the characteristic's value to be the array of bytes that is actually a string.
    //pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
  }else if(doScan){
    BLEDevice::getScan()->start(0);  // this is just eample to start scan after disconnect, most likely there is better way to do it in arduino
  }
  
  delay(3); // Delay a second between loops.
} // End of loop`

tama-ts, 2019-10-25:
Also I would like to say that your code has been of invaluable help in my project!

tama-ts, 2019-10-25:
Also to expand a little - when the line: 

std::string value = pRemoteCharacteristic->readValue();

is not commented I do still receive data from my IMU sensor, it is just not the correct data.

chegewara, 2019-10-26:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteCharacteristic.cpp#L404-L435

but the real problem is that you need to change endian order.
https://devzone.nordicsemi.com/f/nordic-q-a/44630/is-ble-data-sends-in-lsb-first-and-little-endian-byte-order

tama-ts, 2019-10-26:
Thank you for your response! I will test different things today/tomorrow and look into the endian order article.

Meanwhile I have a question that is a little off topic, but I haven't really been able to find any articles on the subject. Once I have programmed my esp through the dev board, will there arise any fatal problems if I take the esp off the dev board and solder my imu sensors straight to the esp as well as a lipo battery for power? If you have any experience with such actions I would love to hear some input. Thank you in advance!

tama-ts, 2019-10-29:
Hello!

Just wanted to throw an update into this thread. We fixed the problem by having the client write it's own IMU sensor data to the remote characteristic, and then having the server read that characteristic and it's own IMU data - and making the required calculations between the two. I am still not sure why the readValue() function in the client was altering the client IMU data, but we did end up finding a work around.

closed 2019-11-04 by tama-ts