nkolban/esp32-snippets #920: BLE_client.ino

tcpipchip, 2019-10-30:
Sir

I am using the BLE_client.ino example only to scan and  connect to my 5 NRF52832 (SERVER), but connection take a long time to get success, about 10 to 15 seconds.

If i use the NORDIC APP i got a imediate connection.

Some tips ?

Thank you so much!

chegewara, 2019-10-30:
Maybe scan and connect code example?

tcpipchip, 2019-10-31:
hi @chegewara 
Thanks for help! 

Now i am getting during the SCAN incomplete packages, for example, my NRF51822 are INDRIYA-TEMP and INDRIYA-SWITCH

When i do the first scan, i can see INDRIYA-TEMP and INDRIYA-SWITCH names.

After the second scan they are showed blank, only two commas (, ,)

Is that a noise problem in my room, that has 2 wifi, a celular, 2 beacon in a 2meter x 2meter room ?

chegewara, 2019-10-31:
I dont think so.
Maybe problem with antenna on esp32? Hard to say.