nkolban/esp32-snippets #921: i need some help really fast

mayank4476843854265972, 2019-11-11:
Arduino: 1.8.10 (Windows 10), Board: "ESP32 Wrover Module, Huge APP (3MB No OTA/1MB SPIFFS), QIO, 80MHz, 921600, None"

C:\Users\ACER PC\Downloads\ARDUINO IDE\arduino-1.8.10\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\ACER PC\Downloads\ARDUINO IDE\arduino-1.8.10\hardware -hardware C:\Users\ACER PC\AppData\Local\Arduino15\packages -tools C:\Users\ACER PC\Downloads\ARDUINO IDE\arduino-1.8.10\tools-builder -tools C:\Users\ACER PC\Downloads\ARDUINO IDE\arduino-1.8.10\hardware\tools\avr -tools C:\Users\ACER PC\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\ACER PC\Downloads\ARDUINO IDE\arduino-1.8.10\libraries -libraries C:\Users\ACER PC\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=huge_app,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=0403_6001 -ide-version=10810 -build-path C:\Users\ACERPC~1\AppData\Local\Temp\arduino_build_119643 -warnings=none -build-cache C:\Users\ACERPC~1\AppData\Local\Temp\arduino_cache_395833 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\ACERPC~1\AppData\Local\Temp\arduino_modified_sketch_705429\CameraWebServer.ino
C:\Users\ACER PC\Downloads\ARDUINO IDE\arduino-1.8.10\arduino-builder -compile -logger=machine -hardware C:\Users\ACER PC\Downloads\ARDUINO IDE\arduino-1.8.10\hardware -hardware C:\Users\ACER PC\AppData\Local\Arduino15\packages -tools C:\Users\ACER PC\Downloads\ARDUINO IDE\arduino-1.8.10\tools-builder -tools C:\Users\ACER PC\Downloads\ARDUINO IDE\arduino-1.8.10\hardware\tools\avr -tools C:\Users\ACER PC\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\ACER PC\Downloads\ARDUINO IDE\arduino-1.8.10\libraries -libraries C:\Users\ACER PC\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=huge_app,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=0403_6001 -ide-version=10810 -build-path C:\Users\ACERPC~1\AppData\Local\Temp\arduino_build_119643 -warnings=none -build-cache C:\Users\ACERPC~1\AppData\Local\Temp\arduino_cache_395833 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\ACER PC\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\ACERPC~1\AppData\Local\Temp\arduino_modified_sketch_705429\CameraWebServer.ino
Using board 'esp32wrover' from platform in folder: C:\Users\ACER
Using core 'esp32' from platform in folder: C:\Users\ACER
Detecting libraries used...
"C:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nghttp" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nvs_flash" "-IC:\\Users\\ACER PC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1 [... cut ...]
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:2a:44:8c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4096.1 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

chegewara, 2019-11-12:
Really fast answer here, you have problem with flashing board. Use google, there is literally hundreds similar issue and answers how to solve it.

closed 2019-11-12 by chegewara

mayank4476843854265972, 2019-11-12:
i know to use google, after 8 hours of searching many issue sound similar i tried them all but none of them seemed to fix my problem so now i turn up to stackoverflow and github for answers when i check my serial monitor and reset pins it states waiting foe download when i start upload it connects to board as well but then this error happens each f time i already tried 5v connection external 
power  connections nothings seems to work

mayank4476843854265972, 2019-11-12:
@chegewara  thanks for replying i have seen your replies in other similar issueses too seems like you are really helpfull and active