nkolban/esp32-snippets #923: How can we reduce memory heap required ?

debendramodi, 2019-11-13:
We are using the following functions and libraries-

Wifi (AP+STA)
MQTT (pubsubclient) with SSL/TLS
BLE

We have released memory used by BT classic.  We find we run into memory heap issues.

We find that this library takes almost 100K heap.  Are there any options to reduce the heap used by this library ?

chegewara, 2019-11-14:
I am currently working on project (cant share code), that is using wifi STA, AWS IoT (mqtt), HTTPS, client, BLE server, RTC and at the end still have about 84kB free memory. Its arduino as component project with platformio.

debendramodi, 2019-11-18:
Hi Chegewara

I wish to continue on the Arduino platform.

I have read in several forums about re-building the Arduino core with GATTC disabled (as we want to use only GATTS ) to help reduce the heap.

I set up the IDF environment and copied the arduino core within the component folder in the blink project.  I deleted the libesp32.a file from the build folder.

The project is built successfully but I do not see the libesp32.a created.

What could I be doing wrong ?

chegewara, 2019-11-18:
Hi, 
this question should be addressed to @me-no-dev 

From my side i can advice you to add this command in setup(), just to make sure it will really release memory in case something is not ok with arduino implementation:
`esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);`

debendramodi, 2019-11-18:
Thanks Chegewara.

I have already disabled Classic BT.  That does not help sufficiently.  Now I am trying to free more heap by disabling GATTC.

On successful build of the Arduino core, without changing any sdk config options (just building the core from the download), I do get the build files in the Build folder.  However the *.a files within the component folder are not changed.
The size of the files are different, For example libbt.a within the component folder is 12 MB while the new build libbt.a in the build folder is 8MB.

Should not the component *.a files be the same as the build files if I am not changing any sdkconfig options ?

On replacing this libbt.a in my arduino IDE, my code does not work.

debendramodi, 2019-11-19:
Thanks Chegewara.  Setup lib_builder again and was able to build the core successfully.  After changing several config options, I am able to save almost 40 KB of heap.  Everything works fine now.

closed 2019-11-19 by debendramodi