nkolban/esp32-snippets #924: Connect two devices parallel or sequential to a BLE Server

speendo, 2019-11-14:
First of all THANK you for maintaining this great project and helping so many new users (like me) how to make their first steps with an ESP32 and Bluetooth.

Using [this project](https://github.com/T-vK/ESP32-BLE-Keyboard) I try to let my ESP32 appear as a BLE Keyboard.

The difficult part is, that I want it to send the same key press events to two clients at the same time.

Is this possible? If yes, what would I have to do?

In case it is not, can I address the two clients sequentially? That means, can I make my ESP32 

1. connect to device A
2. send several key events to device A
3. disconnect device A
4. connect to device B
5. send the same key events to device B
6. repeat

Thank you in advance for your help!

chegewara, 2019-11-15:
Here is basic logic how to allow connect few clients. Then you just need to send notifications to each client. 2nd option is even harder to achieve.

https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_server_multiconnect/BLE_server_multiconnect.ino

speendo, 2019-11-16:
Thank you for your help @chegewara .

So, if I understand you correctly, I will have to remodel this file  
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEServer.cpp

to also have the major properties of the one you mentioned (and authored), right?

I'm not sure if I will be able to do this, as I am very new to BLE. Still, can you give me a hint on how to start? Or is it hopeless for me? :)

Thank you, once more!

chegewara, 2019-11-16:
I dont think you need to change BLEServer at all. You need to start again advertising after connecting to 1 device. Here is code that should send notifications to all connected devices:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.cpp#L510