nkolban/esp32-snippets #926: undefined reference to `app_main'

lucasexequiel, 2019-11-21:
Hi everyone, 
I started a project in Eclipse IDE version 19-03 who works together with ESP-IDF. I want to work with the libraries u8g2 (https://github.com/olikraus/u8g2) and i follow the steps that he propose, building a new file called components and inside call a git clone to download the file u8g2 but when i try compile the console show me the following error. 

/home/lucas/esp/esp-idf/components/esp32/cpu_start.c:528: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
make: *** [/home/lucas/esp/esp-idf/make/project.mk:483: /home/lucas/esp/workspace/SPI_OK/build/SPI_OK.elf] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

I saw this post and I try adding the next code in my proyect but still not work.

void task_test_ST7920(void *ignore) {

	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.clk   = PIN_CLK;
	u8g2_esp32_hal.mosi  = PIN_MOSI;
	u8g2_esp32_hal.cs    = PIN_CS;
	u8g2_esp32_hal.dc    = PIN_DC;
	u8g2_esp32_hal.reset = PIN_RESET;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_t u8g2; 							// a structure which will contain all the data for one display

	u8g2_Setup_st7920_128x64_1(u8g2, rotation, u8x8_byte_8bit_6800mode, uC specific);  // init u8g2 structure

	u8g2_InitDisplay(&u8g2); 				// send init sequence to the display, display is in sleep mode after this,

	u8g2_SetPowerSave(&u8g2, 0); 			// wake up display
	u8g2_ClearBuffer(&u8g2);
	u8g2_DrawBox(&u8g2, 10,20, 20, 30);
	u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	u8g2_DrawStr(&u8g2, 0,15,"Hello World!");
	u8g2_SendBuffer(&u8g2);

	ESP_LOGD(tag, "All done!");

	vTaskDelete(NULL);
}

void app_main(){
	task_test_ST7920;
}

extern "C" {
    void app_main(void);
}
void task_test_ST7920(void *ignore) {

	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.clk   = PIN_CLK;
	u8g2_esp32_hal.mosi  = PIN_MOSI;
	u8g2_esp32_hal.cs    = PIN_CS;
	u8g2_esp32_hal.dc    = PIN_DC;
	u8g2_esp32_hal.reset = PIN_RESET;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_t u8g2; 							// a structure which will contain all the data for one display

	u8g2_Setup_st7920_128x64_1(u8g2, rotation, u8x8_byte_8bit_6800mode, uC specific);  // init u8g2 structure

	u8g2_InitDisplay(&u8g2); 				// send init sequence to the display, display is in sleep mode after this,

	u8g2_SetPowerSave(&u8g2, 0); 			// wake up display
	u8g2_ClearBuffer(&u8g2);
	u8g2_DrawBox(&u8g2, 10,20, 20, 30);
	u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	u8g2_DrawStr(&u8g2, 0,15,"Hello World!");
	u8g2_SendBuffer(&u8g2);

	ESP_LOGD(tag, "All done!");

	vTaskDelete(NULL);
}

void app_main(){
	task_test_ST7920;
}

I am trying to handle a ST7920 screen for my proyect and his comunication is in SPI.

DigiTorus86, 2020-04-12:
Usually this indicates you are missing a CMakeLists.txt file in your main folder.  Should contain something like:
idf_component_register(SRCS "app_main.c"                  
	INCLUDE_DIRS "")

Eonassis, 2023-04-20:
idf_component_register(
SRCS "YOUR_main.c" "u8g2_esp32_hal.c"
INCLUDE_DIRS ".")