nkolban/esp32-snippets #927: not returning to arduino loop

vcmorini, 2019-11-25:
Hello
When I am scanning-->connecting with my BLE device, and a Ticker function, or MQTT message comes into play at the same time, sometimes my ESP32 never goes back to the arduino loop.

It remains only executing the attached Ticker functions, but never back to my loop, until my watchdog (a function attached to another Ticker) is triggered, rebooting the ESP32. 
Could it be a problem of a missing "return" at the end of the attached Ticker function, even if the function return is void? Or maybe a thread/semaphore/priority task issue?
What are your suggestions?
What can I study in order to solve this issue?

Many thanks!

vcmorini, 2019-11-26:
@chegewara I know you have a tight schedule.
I will tag you in hope you can give me your suggestion when you are free, since your opinion weights a lot to me.
Many thanks for your contribution to this library so far!

chegewara, 2019-11-26:
Its hard to say without code or logs.

thomastech, 2019-12-01:
> Or maybe a thread/semaphore/priority task issue?

In case it helps your situation, see the discussion about xSemaphoreTake found here:
https://github.com/nkolban/esp32-snippets/issues/874

- Thomas

vcmorini, 2020-01-05:
> > Or maybe a thread/semaphore/priority task issue?
> 
> In case it helps your situation, see the discussion about xSemaphoreTake found here:
> #874
> 
> * Thomas

Thanks Thomas, this was exactly the issue.

vcmorini, 2020-01-06:
Dear Mr. @chegewara ,
Can this function:
`void notifyCallback(BLERemoteCharacteristic* pBLERemoteCharacteristic, uint8_t* pData, size_t length, bool isNotify)
{ ...` 

Be considered an Interrupt Service Routine (ISR)? I ask this because I am in doubt which type of FreeRTOS functions I should use inside it, i.e. if I should use **fncFromISR** (e.g. https://freertos.org/xSemaphoreTakeFromISR.html) functions or the standard ones.

Many thanks!

chegewara, 2020-01-06:
No, this is regular function, no need to use xxxfromISR in it.