nkolban/esp32-snippets #928: Memory leak on BLEDevice::init() and deinit()

arthurkafer, 2019-11-27:
Using ESP-WROOM-32U, running a simple code to simulate `init` and `deinit` of a BLEDevice.

Compile settings:
- Board: ESP32 Dev Module;
- Upload Speed: 921600;
- Flash Frequency: 80MHz;
- Flash Mode: QIO;
- Partition Scheme: Default;
- Core Debug Level: Verbose;
- PSRAM: Disabled;

```
#include <BLEDevice.h>

void setup() {
  // put your setup code here, to run once:
    Serial.begin(115200);
    BLEDevice::init("testeaqi");
}

void loop() {
    Serial.println(ESP.getFreeHeap());
    if (BLEDevice::getInitialized() == true){
        Serial.println("BLE_OFF()");
        BLEDevice::deinit(false);
    }
    else {
        Serial.println("NEW_INIT()");
        BLEDevice::init("testeasd");
    }
}
```

Here are the logs:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9296
load:0x40080400,len:6364
entry 0x400806a4
184740
BLE_OFF()
268236
NEW_INIT()
181468
BLE_OFF()
264960
NEW_INIT()
178180
BLE_OFF()
261536
NEW_INIT()
174760
BLE_OFF()
258248
NEW_INIT()
171476
BLE_OFF()
254828
NEW_INIT()
168072
BLE_OFF()
251552
NEW_INIT()
164800
BLE_OFF()
248424
NEW_INIT()
161664
BLE_OFF()
245152
NEW_INIT()
158400
BLE_OFF()
241880
NEW_INIT()
155116
BLE_OFF()
238460
NEW_INIT()
151712
BLE_OFF()
235332
NEW_INIT()
148584
BLE_OFF()
231912
NEW_INIT()
145164
BLE_OFF()
228636
NEW_INIT()
141912
BLE_OFF()
225368
NEW_INIT()
138756
BLE_OFF()
222244
NEW_INIT()
135484
BLE_OFF()
218972
NEW_INIT()
132216
BLE_OFF()
220036
NEW_INIT()
128844
BLE_OFF()
212428
NEW_INIT()
125664
BLE_OFF()
209156
NEW_INIT()
122276
BLE_OFF()
205736
NEW_INIT()
119008
BLE_OFF()
202464
NEW_INIT()
115736
BLE_OFF()
199336
NEW_INIT()
112584
BLE_OFF()
196060
NEW_INIT()
109300
BLE_OFF()
192788
NEW_INIT()
105884
BLE_OFF()
193476
NEW_INIT()
102752
BLE_OFF()
186240
NEW_INIT()
99504
BLE_OFF()
182820
NEW_INIT()
96092
BLE_OFF()
179548
NEW_INIT()
92820
BLE_OFF()
176424
NEW_INIT()
89680
BLE_OFF()
173152
NEW_INIT()
86412
BLE_OFF()
169880
NEW_INIT()
83136
BLE_OFF()
166460
NEW_INIT()
79720
BLE_OFF()
163332
NEW_INIT()
76588
BLE_OFF()
159916
NEW_INIT()
73164
BLE_OFF()
161128
NEW_INIT()
69888
BLE_OFF()
153372
NEW_INIT()
66756
BLE_OFF()
150240
NEW_INIT()
63476
BLE_OFF()
146968
NEW_INIT()
60204
BLE_OFF()
148036
NEW_INIT()
56800
BLE_OFF()
140424
NEW_INIT()
53536
BLE_OFF()
137152
NEW_INIT()
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c46c  PS      : 0x00060b30  A0      : 0x8011ab82  A1      : 0x3ffd7d30  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000001c  A5      : 0x00000000  
A6      : 0x4018844c  A7      : 0x00000001  A8      : 0x800dc11e  A9      : 0x3ffd7cd0  
A10     : 0x3ffe3e60  A11     : 0x00000044  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00001800  A15     : 0x00000084  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4000c46c:0x3ffd7d30 0x4011ab7f:0x3ffd7d40 0x4011b455:0x3ffd7d60 0x400f6279:0x3ffd7d80 0x400f62b3:0x3ffd7da0 0x40092da5:0x3ffd7dd0

Rebooting...

```

The backtrace results in this: 

```
0x4011ab7f: bta_ar_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/ar/bta_ar.c line 72

Decoding stack results
0x4011b455: bta_sys_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/sys/bta_sys_main.c line 186
0x400f6279: btu_task_start_up at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 284
0x400f62b3: btu_task_thread_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 226
0x40092da5: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
0x4011ab7f: bta_ar_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/ar/bta_ar.c line 72
0x4011b455: bta_sys_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/sys/bta_sys_main.c line 186
0x400f6279: btu_task_start_up at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 284
0x400f62b3: btu_task_thread_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 226
0x40092da5: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

Every single iteration of the init/deinit loop is leaking ~4k, and I could'nt find any other way to fix this. My initial problem is based on this memory leak, I think you could help me with this.

Thanks in advance

arthurkafer, 2019-11-27:
I saw something about the memory leak on #900, but in my example I'm just calling `init` and `deinit`. Other possible issue is [this one](https://github.com/espressif/arduino-esp32/issues/3389), but I couldn't change anything on the raw library to make the results better.

TheNitek, 2020-02-18:
I'm seeing similar behaviour: I'm alternating between WiFi and BLE (to make sure enough heap is available in both situations). After every BLE-Iteration the heap decreases until the ESP crashes.

chegewara, 2020-02-22:
I tested with code from OP and here is result:
```
15:47:41.341 -> ets Jun  8 2016 00:22:57
15:47:41.341 -> 
15:47:41.341 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:47:41.341 -> configsip: 0, SPIWP:0xee
15:47:41.341 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:47:41.341 -> mode:DIO, clock div:1
15:47:41.341 -> load:0x3fff0018,len:4
15:47:41.341 -> load:0x3fff001c,len:1216
15:47:41.341 -> ho 0 tail 12 room 4
15:47:41.341 -> load:0x40078000,len:9720
15:47:41.341 -> ho 0 tail 12 room 4
15:47:41.341 -> load:0x40080400,len:6352
15:47:41.341 -> entry 0x400806b8
15:47:42.367 -> 189016
15:47:42.367 -> BLE_OFF()
15:47:42.566 -> 274860
15:47:42.566 -> NEW_INIT()
15:47:43.131 -> 189024
15:47:43.131 -> BLE_OFF()
15:47:43.364 -> 274868
15:47:43.364 -> NEW_INIT()
15:47:43.929 -> 189024
15:47:43.929 -> BLE_OFF()
15:47:44.127 -> 274712
15:47:44.127 -> NEW_INIT()
15:47:44.693 -> 188876
15:47:44.693 -> BLE_OFF()
15:47:44.892 -> 274716
15:47:44.892 -> NEW_INIT()
15:47:45.491 -> 188876
15:47:45.491 -> BLE_OFF()
15:47:45.691 -> 274716
15:47:45.691 -> NEW_INIT()
15:47:46.256 -> 188888
15:47:46.256 -> BLE_OFF()
15:47:46.456 -> 274712
15:47:46.456 -> NEW_INIT()
15:47:47.056 -> 188868
15:47:47.056 -> BLE_OFF()
15:47:47.255 -> 274716
15:47:47.255 -> NEW_INIT()
15:47:47.821 -> 188888
15:47:47.821 -> BLE_OFF()
15:47:48.054 -> 274712
15:47:48.054 -> NEW_INIT()
15:47:48.619 -> 188864
15:47:48.619 -> BLE_OFF()
15:47:48.819 -> 274716
15:47:48.819 -> NEW_INIT()
15:47:49.384 -> 188884
15:47:49.384 -> BLE_OFF()
15:47:49.617 -> 274712
15:47:49.617 -> NEW_INIT()
15:47:50.184 -> 188868
15:47:50.184 -> BLE_OFF()
15:47:50.383 -> 274716
15:47:50.383 -> NEW_INIT()
15:47:50.948 -> 188884
15:47:50.948 -> BLE_OFF()
15:47:51.181 -> 274712
15:47:51.181 -> NEW_INIT()
15:47:51.713 -> 188868
15:47:51.713 -> BLE_OFF()
15:47:51.980 -> 274716
15:47:51.980 -> NEW_INIT()
15:47:52.545 -> 188884
15:47:52.545 -> BLE_OFF()
15:47:52.744 -> 274712
15:47:52.744 -> NEW_INIT()
15:47:53.309 -> 188868
15:47:53.309 -> BLE_OFF()
15:47:53.542 -> 274716
15:47:53.542 -> NEW_INIT()
15:47:54.107 -> 188884
15:47:54.107 -> BLE_OFF()
15:47:54.306 -> 274712
15:47:54.306 -> NEW_INIT()
15:47:54.904 -> 188868
15:47:54.904 -> BLE_OFF()
15:47:55.104 -> 274716
15:47:55.104 -> NEW_INIT()
15:47:55.702 -> 188884
15:47:55.702 -> BLE_OFF()
15:47:55.934 -> 274712
15:47:55.934 -> NEW_INIT()
15:47:56.500 -> 188872
15:47:56.500 -> BLE_OFF()
15:47:56.732 -> 274720
15:47:56.732 -> NEW_INIT()
15:47:57.298 -> 188876
15:47:57.298 -> BLE_OFF()
15:47:57.497 -> 274712
15:47:57.497 -> NEW_INIT()
15:47:58.062 -> 188880
15:47:58.062 -> BLE_OFF()
15:47:58.261 -> 274716
15:47:58.261 -> NEW_INIT()
15:47:58.859 -> 188880
15:47:58.859 -> BLE_OFF()
15:47:59.058 -> 274712
15:47:59.058 -> NEW_INIT()
15:47:59.691 -> 188872
15:47:59.691 -> BLE_OFF()
15:47:59.891 -> 274716
15:47:59.891 -> NEW_INIT()
15:48:00.459 -> 188876
15:48:00.459 -> BLE_OFF()
15:48:00.691 -> 274712
15:48:00.691 -> NEW_INIT()
15:48:01.256 -> 188880
15:48:01.256 -> BLE_OFF()
15:48:01.455 -> 274716
15:48:01.455 -> NEW_INIT()
15:48:02.019 -> 188876
15:48:02.019 -> BLE_OFF()
15:48:02.218 -> 274712
15:48:02.218 -> NEW_INIT()
15:48:02.817 -> 188880
15:48:02.817 -> BLE_OFF()
15:48:03.084 -> 274716
15:48:03.084 -> NEW_INIT()
15:48:03.649 -> 188880
15:48:03.649 -> BLE_OFF()
15:48:03.848 -> 274712
15:48:03.848 -> NEW_INIT()
15:48:04.414 -> 188876
15:48:04.414 -> BLE_OFF()
15:48:04.646 -> 274716
15:48:04.646 -> NEW_INIT()
15:48:05.211 -> 188880
15:48:05.211 -> BLE_OFF()
15:48:05.411 -> 274712
15:48:05.411 -> NEW_INIT()
15:48:06.009 -> 188876
15:48:06.009 -> BLE_OFF()
15:48:06.208 -> 274716
15:48:06.208 -> NEW_INIT()
15:48:06.772 -> 188880
15:48:06.772 -> BLE_OFF()
15:48:07.005 -> 274712
15:48:07.005 -> NEW_INIT()
15:48:07.602 -> 188880
15:48:07.602 -> BLE_OFF()
15:48:07.801 -> 274716
15:48:07.801 -> NEW_INIT()
15:48:08.366 -> 188876
15:48:08.366 -> BLE_OFF()
15:48:08.599 -> 274712
15:48:08.599 -> NEW_INIT()
15:48:09.162 -> 188880
15:48:09.162 -> BLE_OFF()
15:48:09.362 -> 274716
15:48:09.362 -> NEW_INIT()
15:48:09.926 -> 188876
15:48:09.926 -> BLE_OFF()
15:48:10.158 -> 274712
15:48:10.158 -> NEW_INIT()
15:48:10.788 -> 188872
15:48:10.788 -> BLE_OFF()
15:48:10.987 -> 274716
15:48:10.987 -> NEW_INIT()
15:48:11.586 -> 188880
15:48:11.586 -> BLE_OFF()
15:48:11.784 -> 274712
15:48:11.784 -> NEW_INIT()
15:48:12.348 -> 188876
15:48:12.348 -> BLE_OFF()
15:48:12.547 -> 274716
15:48:12.547 -> NEW_INIT()
15:48:13.146 -> 188880
15:48:13.146 -> BLE_OFF()
15:48:13.345 -> 274712
15:48:13.345 -> NEW_INIT()
15:48:13.908 -> 188876
15:48:13.908 -> BLE_OFF()
15:48:14.174 -> 274716
15:48:14.174 -> NEW_INIT()
15:48:14.738 -> 188884
15:48:14.738 -> BLE_OFF()
15:48:14.970 -> 274716
15:48:14.970 -> NEW_INIT()
15:48:15.535 -> 188884
15:48:15.535 -> BLE_OFF()
15:48:15.735 -> 274716
```

Not even a single byte leaking anymore.

TheNitek, 2020-02-22:
Sorry for the confusion: I got the same problem using the BLEDevice implementation in arduino-esp32.  During my debugging/gooleing session I must have gotten confused and didn't realize this was a different repo.

closed 2020-05-11 by arthurkafer

peter1a, 2020-10-02:
Anyone found solution for ble init / deinit leaking problem?

arthurkafer, 2020-10-13:
@peter1a It is a repository problem, you should use the library from [here](https://github.com/espressif/arduino-esp32)

libasoles, 2020-10-16:
I don't get it. If I go to [that link](https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE
) I see this.
> The actual source of the project which is being maintained can be found here:

> https://github.com/nkolban/esp32-snippets

So here we are again.

chegewara, 2020-10-16:
When you check that README.md is 2 years old and has not been updated.
Currently this library is no longer maintained
https://github.com/nkolban/esp32-snippets/issues/813

and few more people, including me, is trying to answer questions and help with issues.

libasoles, 2020-10-16:
Ok, I see. Thanks for that link. 

I think it would be important to update this README then:
https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE

 And thanks for your efforts.