nkolban/esp32-snippets #929: BLE srv write scan response Warning partial data write

zubize, 2019-11-27:
Hi, 
I'm developig a firmware test on ESP32 and I use the BLE library. 
I made the service automation with 12 characteristic but when I start the advertising object give me the warning message **"BT_BTM: BTM_BleWriteScanRsp, Partial data write into ADV"**.
Into the ESP32 stack there is a comment referred to 31 byte size pack but I don't understand where is a problem and if increasing this size the problem disappears. 
However when I scan the service by phone the information are partial.
Can anyone help me understand the problem?

thanks

chegewara, 2019-11-28:
The problem is you cant increase it, because its bluetooth standard. Advertising and scan response packets can have max 31 bytes each, period.

zubize, 2019-11-29:
Hi increase but the result doesn't change. Hi think that this warning depending to flags enabling into adv. The mask returns is 0x20 referred to BTM_BLE_AD_BIT_INT_RANGE, but I don't understand where and hot to use it.

thanks

chegewara, 2019-11-29:
If i understand correctly 0x20 is `Service Data - 32-bit UUID`, but if its `BTM_BLE_AD_BIT_INT_RANGE`, then you have this `Slave Connection Interval Range` set wrong values, which is 0x12 in advertising packet.

zubize, 2019-11-29:
I think that the warning message tricked me. My promblem is that  when the phone app scan services available it lists wrong the characteristic, the first 2 are correct but the 3° not and into a service are present 12 characteristic. I used the UUID standard service AUTOMATION_IO and characteristic Digital end Analog, and list the fist 3 characteristics only.

[image]
Attach

zubize, 2019-11-29:
[image]

chegewara, 2019-11-29:
I am suspecting that service handles are depleted, try to create service with value 20 or more. 
Also its possible you have issue with your code.

zubize, 2019-11-29:
I treid to insert 10 characteristics in a other custom service but on the app i see 7 only ...

the BLEUUID(true) make a new UUID ...

>     // DBG: added to verify the service characteristics
>     pService = mpServer->createService(BLEUUID(UUID_SERVICE));
>     mServices.insert(ItemService(pService->getUUID().toString(), pService));
>     if (pService) {
>         for (int i=0; i<10; i++) {
> 
>             BLECharacteristic *pCharacteristic = pService->createCharacteristic(BLEUUID(true), BLECharacteristic::PROPERTY_READ);
>             ESP_LOGI(LOG_TAG, "Added test Characteristic: %s, handle: %x", pCharacteristic->getUUID().toString().c_str()
>                                                                          , pCharacteristic->getHandle());
>         }
>     }

this i the log result

> I (18482) BLESRV: Added test Characteristic: fb349b5f-8000-0080-0010-000020212400, handle: ffff
> I (18482) BLESRV: Added test Characteristic: fb349b5f-8000-0080-0010-000020212401, handle: ffff
> I (18502) BLESRV: Added test Characteristic: fb349b5f-8000-0080-0010-000020212402, handle: ffff
> I (18502) BLESRV: Added test Characteristic: fb349b5f-8000-0080-0010-000020212403, handle: ffff
> I (18512) BLESRV: Added test Characteristic: fb349b5f-8000-0080-0010-000020212404, handle: ffff
> I (18522) BLESRV: Added test Characteristic: fb349b5f-8000-0080-0010-000020212405, handle: ffff
> I (18532) BLESRV: Added test Characteristic: fb349b5f-8000-0080-0010-000020212406, handle: ffff
> I (18542) BLESRV: Added test Characteristic: fb349b5f-8000-0080-0010-000020212407, handle: ffff
> I (18552) BLESRV: Added test Characteristic: fb349b5f-8000-0080-0010-000020212408, handle: ffff
> I (18562) BLESRV: Added test Characteristic: fb349b5f-8000-0080-0010-000020212409, handle: ffff

chegewara, 2019-11-29:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEServer.h#L67

zubize, 2019-11-29:
Ok now it works, I have many characteristics. I increase a 50 and it works.

Thanks a lot!!

closed 2019-11-29 by zubize