nkolban/esp32-snippets #932: Changing Scan Time From Second To Millisecond

NimaaZx, 2019-12-07:
Hello.
I wanted to change scan time property from Second to Millisecond. There is scanTime in BLEScan example File in Arduino library. 

```
int scanTime = 5 ; //In seconds

BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
```
It is used there. First It accepted Integer but I changed the these two Function in .h library Files + .CPP file to Float. 

```
bool  start(float duration, void (*scanCompleteCB)(BLEScanResults), bool is_continue = false);
	BLEScanResults start(float duration, bool is_continue = false);
..

```

then change int to float in the main arduino file.
`float scanTime = 0.5 ; //In seconds - means 500 ms`

File is uploaded in Esp32 with no error ,then Receive first BLE signal but is not received more data .

is there anybody to know what do i do for this changing ?

chegewara, 2019-12-07:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv426esp_ble_gap_start_scanning8uint32_t

NimaaZx, 2019-12-08:
> https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv426esp_ble_gap_start_scanning8uint32_t

Thank you. I  Changed the code to 

```
bool BLEScan::start(**float duration**, void (*scanCompleteCB)(BLEScanResults), bool is_continue) {
	ESP_LOGD(LOG_TAG, ">> start(duration=%.1f)", duration);

	m_semaphoreScanEnd.take(std::string("start"));
	m_scanCompleteCB = scanCompleteCB;                  // Save the callback to be invoked when the scan completes.

	//  if we are connecting to devices that are advertising even after being connected, multiconnecting peripherals
	//  then we should not clear map or we will connect the same device few times
	if(!is_continue) {  
		for(auto _dev : m_scanResults.m_vectorAdvertisedDevices){
			delete _dev.second;
		}
		m_scanResults.m_vectorAdvertisedDevices.clear();
	}

	esp_err_t errRc = ::esp_ble_gap_set_scan_params(&m_scan_params);

	if (errRc != ESP_OK) {
		ESP_LOGE(LOG_TAG, "esp_ble_gap_set_scan_params: err: %d, text: %s", errRc, GeneralUtils::errorToString(errRc));
		m_semaphoreScanEnd.give();
		return false;
	}

	errRc = ::**esp_ble_gap_start_scanning(duration)**;

	if (errRc != ESP_OK) {
		ESP_LOGE(LOG_TAG, "esp_ble_gap_start_scanning: err: %.1f, text: %s", errRc, GeneralUtils::errorToString(errRc));
		m_semaphoreScanEnd.give();
		return false;
	}

	m_stopped = false;

	ESP_LOGD(LOG_TAG, "<< start()");
	return true;
} // start
```

but is not working . received first BLE packet until I turn off and turn on again my BLE beacon and again ESP32 receive second BLE Packet. it can not receive Packet Continuously. I think that Function return false value but app is not crashed.

I read this topic . 
https://esp32.com/viewtopic.php?t=2291
interesting but  I think window-size and Interval is far different from Nkolban or that Image said. because this app is not working than it expected. IF I change scanTime to 10 Second . It receives ble packet each every 10 Second. So window size and interval is useless. I changed them but nothing special happened.

chegewara, 2019-12-08:
You can setup window and interval values in your code. Try to set very short values and very long values and compare results.