nkolban/esp32-snippets #933: SPP via BLE

squonk11, 2019-12-22:
Hello,
although this repository seems to be "out of service" I post my question here hoping that someone can point me to the right approach.
I would like to setup something similar to SPP in classic Bluetooth but now with BLE. A PC (or smartphone) app will connect to the ESP32 via BLE and all data shall be tunneled to and from a UART.
Is there somewhere a useful example to start with?
Is it possible to use the cpp classes without Arduino?
Where is the latest version of this repository - I see that @chegewara also maintained a copy of this one?
Any hint would be helpful.
Kind regards

chegewara, 2019-12-22:
Hi,
i think the latest useful version is in this repository, but maybe someone forked it and did some good changes in code.
I know you dont want to use arduino, but this code is good start:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_uart/BLE_uart.ino

It can be easy used with esp-idf and this library, only thing you have to do is to add UART code that will read from UART and send notifications and handle onWrite() callback and send data to UART.

squonk11, 2019-12-23:
Hello,
thank you for your kind reply. Good to know that these sources are the most recent ones. I tried to compile them against my code which includes the latest esp-idf master branch. Unfortunately here I am running into compilation errors like `FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared`. I think this is due to I am compiling against esp-idf version 4.1 and the actual sources of this repository are actually designed for an older version.
I will check in more detail during the next days.

chegewara, 2019-12-23:
There is issue on this repository, you can search for it. (ringbuf_type_t)

squonk11, 2019-12-30:
now I fixed the `ringbuf_type_t` issue by replacing it with `RingbufferType_t`.
There was also an issue with `esp_ble_gap_update_whitelist` due to a missing parameter in the function call. I fixed this one according to [this hint](https://github.com/nkolban/esp32-snippets/issues/472#issuecomment-501418988)
Is there a plan to update this library in order to stay in line with the latest esp-idf master?

squonk11, 2019-12-30:
Now I have setup everything according to the [example.](https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_uart/BLE_uart.ino) When starting the BLE server I get the error message: 

> E (1869) BLEDevice: esp_bt_controller_enable: rc=258 Unknown ESP_ERR error

Do you have an idea what I am missing?

Best regards

chegewara, 2019-12-30:
> Is there a plan to update this library in order to stay in line with the latest esp-idf master?

You can fork repo and apply required changes in code and eventually create PR with new branch.
Remember that esp-idf v3.3 is still last stable release.

BLE server I get the error message:
> 
> > E (1869) BLEDevice: esp_bt_controller_enable: rc=258 Unknown ESP_ERR error
> 
> Do you have an idea what I am missing?

Honestly i have no idea why it is not working for you, because i am using this code in pretty big project to connect wit RPi and i dont have problem.
For some reason you have problems with initializing BLE, maybe short on heap?

squonk11, 2019-12-31:
> Remember that esp-idf v3.3 is still last stable release.

I think there is already a version 4.0 of esp-idf [available](https://github.com/espressif/esp-idf/tree/release/v4.0). But I agree that this repository is mainly used in Arduino environments and the Arduino port of it is still based on v3.3. So, the little differences in the API probably can be handeled by those people (like me) who want to use it with esp-idf.

> E (1869) BLEDevice: esp_bt_controller_enable: rc=258 Unknown ESP_ERR error

in the meantime I found a solution for this [here](https://github.com/espressif/esp-idf/issues/405). Nevertheless: thank you for your fast answer.

bbinet, 2020-01-16:
@squonk11 I've also noticed the following BLE-UART example which does not use arduino libs (esp-idf only):
https://github.com/pcbreflux/espressif/tree/master/esp32/app/ESP32_ble_UART
See also related video: https://www.youtube.com/watch?v=FvIywveUVZw

bbinet, 2020-01-16:
There is also the BLE SPP client/server examples in esp-idf sdk:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_spp_client
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_spp_server

chegewara, 2020-01-16:
> There is also the BLE SPP client/server examples in esp-idf sdk:
> https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_spp_client
> https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_spp_server

Problem with espressif example is it is not using Nordic Semiconductor BLE UART, it is using random 16-bit UUIDs, which wont pass bluetooth certification in commercial product.

bbinet, 2020-01-16:
@chegewara Can't we just update the uuids in the esp-idf ble_ssp examples to be compatible with Nordic Semiconductor BLE UART?

chegewara, 2020-01-17:
Its not only about UUIDs, espressif is also using completely different characteristics "design".