nkolban/esp32-snippets #935: Unable to retrieve iBeacon UUID

Kevistar, 2019-12-30:
Hi there,

I am working on a project which involves using the ESP32 BLE Scanner. I edit and uploaded the sample code of BLE Scanner in the Arduino IDE library but I was unable to retrieve the UUIDs of iBeacon. The serial monitor always displays "N/a".However, I was able to retrieve the UUIDs when using a normal android BLE scanner. Is there anything missing in my code? Here is my edited code :

`
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEUUID.h>
#include <BLEAdvertisedDevice.h>
BLEScan* pBLEScan;
int scanTime = 5; //In seconds
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks 
{
    void onResult(BLEAdvertisedDevice advertisedDevice) 
    {
      //Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
      Serial.print("Name :");
      
      if (advertisedDevice.haveName())
      {
      Serial.println(advertisedDevice.getName().c_str());
      }
      else
      {
        Serial.println("N/a");
      }
      
      Serial.print("RSSI :");
      Serial.println(advertisedDevice.getRSSI());

      
      Serial.print("UUID :");
      
      if (advertisedDevice.haveServiceDataUUID())
     { 
       Serial.println(advertisedDevice.getServiceDataUUID().toString().c_str());
     }
     else
     {
       Serial.println("N/a");
     }
      
      Serial.println("");  
    }
    
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");
  //BLEDevice::init("");
}

void loop() {
  //Serial.println("Scanning...");
  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  
  delay(1000);
}
`

Kevistar, 2019-12-30:
Sorry if there is any wrong format of posting as this is my first time in asking a question.

chegewara, 2019-12-31:
@Kevistar iBeacon UUID is not advertising the same way as normal UUIDs and you cant retrieve it this way.
My suggestion is to get raw advertising data and use BLEBeacon class to create new object, hen use raw advertising data to set beacon data:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEBeacon.cpp#L59

and you should be able to read UUID.

Kevistar, 2019-12-31:
I will try to work on it first, will inform you of the progress after trying the suggested methods.

Kevistar, 2019-12-31:
@chegewara how should I extract the raw advertising data of the beacons? I was thinking of using the getData() function but no idea of how to write it in a proper format in Arduino.

chegewara, 2019-12-31:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L512

Kevistar, 2019-12-31:
I extracted the raw advertising data (not sure is correct or not) and attempted to set the beacon data but encountered error. Here is the part to display beacon uuid:

`         
          uint8_t*payload= advertisedDevice.getPayload();   //get advertised and scanning response 
          id.setData()= payload;   //ERROR : NO matching function for call to 'BLEBeacon::setData()'
          Serial.print("UUID :");
          Serial.println(id.getProximityUUID().toString().c_str());
`
Is there anything missing or the method of obtaining raw advertising data is wrong, which resulted in the mentioned error.

chegewara, 2019-12-31:
This is function in class object:
`void BLEBeacon::setData(std::string data)`
you need to instantiate object and use function like usual.

Kevistar, 2020-01-02:
I remodified to code and manage to compile and upload it. Here is the new code
`/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEUUID.h>
#include <BLEAdvertisedDevice.h>
#include <BLEBeacon.h>
BLEScan* pBLEScan;
BLEBeacon id;

int scanTime = 5; //In seconds

          
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks 
{
    void onResult(BLEAdvertisedDevice advertisedDevice) 
    {
          //Print Name
          Serial.print("Name :");
          if (advertisedDevice.haveName())
          {
          Serial.println(advertisedDevice.getName().c_str());
          }
          else
          {
            Serial.println("N/a");
          }

          
          //Print RSSI
          Serial.print("RSSI :");
          Serial.println(advertisedDevice.getRSSI());   

          //Print UUID
          uint8_t* payload = advertisedDevice.getPayload();
          String info = String((char*)payload);
          id.setData(info.c_str());
          Serial.print("UUID :");
          Serial.println(id.getProximityUUID().toString().c_str());
       
          Serial.println("");
    }
    
    
    
};
       
void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");
}

void loop() 
{
  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!"); 
  delay(1000);
}`

But the results that I obtained is shown below:
Name :Ibeacon_1
RSSI :-63
UUID :00000000-0000-0000-0000-000000000000

It seems like there are mistakes in extracting information from the payload. Are there methods for me to get the correct format of advertising data from the payload.

chegewara, 2020-01-02:
Check `info.length()`, should equal 31.

Kevistar, 2020-01-02:
No, it is only 6. Does that mean the conversion of the variable "payload" has been performed wrongly?

chegewara, 2020-01-02:
> No, it is only 6. Does that mean the conversion of the variable "payload" has been performed wrongly?

Yes, because somewhere in middle of payload is 0x0 which String(payload) is treating like end of string. You have to use different way to retrieve data. Try `snprintf` or `sprintf`.

Kevistar, 2020-01-02:
Here is the modified code with sprintf
`    //Print UUID
      const uint8_t* payload = advertisedDevice.getPayload();
      sprintf(info,"%u",payload);
      Serial.print("info:");
      Serial.println(info);
      Serial.println("");`

In the Serial monitor I get the below responds:
Name :iBeacon1
RSSI :-59
info:1073600256

Are the values of info represent the information needed?

chegewara, 2020-01-02:
Now when you use sprintf with rest of your code it should works:
```
      id.setData(info.c_str());
      Serial.print("UUID :");
      Serial.println(id.getProximityUUID().toString().c_str());
   
      Serial.println("");
```
If it wont give you expected results then you probably will need to modify this function a bit, which to be honest, should looks differently:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEBeacon.cpp#L59-L65

```
void BLEBeacon::setData(uint8_t* data) {
	memcpy(&m_beaconData, data.data(), sizeof(m_beaconData));
}
```
Remember to change function definition in BLEBeacon.h

Kevistar, 2020-01-03:
Using the sprintf function does produce results. However the uuid display constantly changes. I also tried to modify the mentioned function. But there is an error during the compile process:

ERROR MESSAGE:
request for member 'data' in 'data', which is of non-class type 'uint8_t* {aka unsigned char*}'

  memcpy(&m_beaconData, data.data(), sizeof(m_beaconData));

chegewara, 2020-01-03:
Yes, sorry for that (copy/paste error):
`memcpy(&m_beaconData, data, sizeof(m_beaconData));`

Kevistar, 2020-01-04:
I followed your suggestion and its working. The UUID has been displayed but is different from the UUID detected by the android BLE scanner app. Maybe this is due to the changing of format for the setData function. I think I can work out the rest from here. Thanks for the help!

chegewara, 2020-01-04:
Im guessing that UUID bytes order is reverted, you can get UUID you have and use this function to create new UUID with reverted order:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEUUID.h#L24

Then you can do whatever you want, filter etc

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEUUID.h#L28

closed 2020-01-06 by Kevistar

ADCETIAN, 2020-08-09:
> Here is the modified code with sprintf
> ` //Print UUID const uint8_t* payload = advertisedDevice.getPayload(); sprintf(info,"%u",payload); Serial.print("info:"); Serial.println(info); [...]

> I followed your suggestion and its working. The UUID has been displayed but is different from the UUID detected by the android BLE scanner app. Maybe this is due to the changing of format for the setData function. I think I can work out the rest from here. Still, just one last question before closing the issues, is it still possible to implement a uuid filter after such modifications have been made?

sorry sir but still i am getting same things
Name :pBeacon
RSSI :-69
info:1065398308

00000000-0000-0000-0000-000000000000

ADCETIAN, 2020-08-09:
> Yes, sorry for that (copy/paste error):
> `memcpy(&m_beaconData, data, sizeof(m_beaconData));`

sorry sir but still i am getting same things
Name :pBeacon
RSSI :-69
info:1065398308

00000000-0000-0000-0000-000000000000

dzungpv, 2020-08-24:
@Kevistar I have the same problem. Edit code can produce UUID but it not same with data in Ibeacon code. Can you share complete code?

thedjenny, 2020-08-26:
> Hi there,
>
> I am working on a project which involves using the ESP32 BLE Scanner. I edit and uploaded the sample code of BLE Scanner in the Arduino IDE library but I was unable to retrieve the UUIDs [...]

hello sir i face the same problem as you once did , 
i have the ibeacon simulator on my android app serving and can detect with any other beacon scanner but i can't with my esp32 and i didn't found a clue with this ServiceUUID and UUID can you help me please?

thedjenny, 2020-08-26:
> Im guessing that UUID bytes order is reverted, you can get UUID you have and use this function to create new UUID with reverted order: [...]

hello sir i am facing troubles with the same problem can you help me please?

monday8, 2022-09-03:
```
BLEBeacon id;
uint8_t *payload = advertisedDevice.getPayload();
id.setData(payload);

String m_uuid = id.getProximityUUID().to128().toString().c_str();
```