nkolban/esp32-snippets #937: Running BLE_server on core 1 affects timing of measurements on core 0

RobWilks, 2020-01-03:
I am seeking help with a program to make a high frequency measurement and then report it out by a BLE connection. My current problem is that measurements are intermittently delayed.
The measurement code is running on core 0 at 1kHz; the duration of each measurement is 400msec with a 80MHz esp32 CPU clock frequency.
The BLE_server runs on core 1, advertising, connecting and disconnecting correctly. It is a direct copy of the example BLE_server in Nicks excellent library.
The timing of core 0 measurements is intermittently delayed by up to 600 msec every 40 or so measurements as measured by a scope attached to a GPIO port. The delays do not occur when the core 1 task is an empty while (true) loop (no BLE comms).
I had wrongly assumed that the core 1 BLEcomms task would not affect the measurement task on core 0. What interrupts on core 0 might cause such a delay? 
Note core0 WDT is disabled.
I would appreciate any ideas of how I might debug this, and even better, how I might improve the code!  A few details follow below.

Many thanks

Rob

===============================

The tasks are created as follows.  There are only two.  

`  disableCore0WDT(); // source https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal.h#L76-L91
  // see epic thread https://github.com/espressif/arduino-esp32/issues/595
  
  //create a task that will be executed in the measureVibration() function, with priority 2 and executed on core 0
  xTaskCreatePinnedToCore(
  measureVibration,   /* Task function. */
  "vibeTask",     /* name of task. */
  10000,       /* Stack size of task */
  NULL,        /* parameter of the task */
  2,           /* priority of the task */
  &vibeTask,      /* Task handle to keep track of created task */
  0);          /* pin task to core 0 */

  
  //create a task that will be executed in the BLEComms() function, with priority 1 and executed on core 1
  xTaskCreatePinnedToCore(
  BLEComms,   /* Task function. */
  "commsTask",     /* name of task. */
  28800,       /* Stack size of task */
  NULL,        /* parameter of the task */
  1,           /* priority of the task */
  &commsTask,      /* Task handle to keep track of created task */
  1);          /* pin task to core 1 */`

A timer is set-up to create 1kHz ticks communicated to the measurement task by semaphore

`////////////////////////////////////// onTimer //////////////////////////////////////////

void IRAM_ATTR onTimer(){
	// Increment the counter and set the time of ISR
	portENTER_CRITICAL_ISR(&timerMux);
	isrCounter++;
	lastIsrAt = millis();
	portEXIT_CRITICAL_ISR(&timerMux);
	
	
	// Give a semaphore that we can check in the loop
	xSemaphoreGiveFromISR(timerSemaphore, NULL);
	// It is safe to use digitalRead/Write here if you want to toggle an output

}
////////////////////////////////////// initTime //////////////////////////////////////////

void initTime(uint32_t period)
// check type for period
{
  // Create semaphore to inform us when the timer has fired
  timerSemaphore = xSemaphoreCreateBinary();

  // Use 1st timer of 4 (counted from zero).
  // Set 80 divider for prescaler (see ESP32 Technical Reference Manual for more
  // info).
  timer = timerBegin(0, 80, true);

  // Attach onTimer function to our timer.
  timerAttachInterrupt(timer, &onTimer, true);

  // Set alarm to call onTimer function every second (value in microseconds).
  // Repeat the alarm (third parameter)
  timerAlarmWrite(timer, period, true);

  // Start an alarm
  timerAlarmEnable(timer);
}`

chegewara, 2020-01-04:
My suggestion is to revert code pinned to core, because most likely bluetooth is running on core 0 (even if you create and run BLE library on core 1). 
Remember that BLE library is only wrapper for esp-idf and bluedroid driver.

RobWilks, 2020-01-04:
@chegewara 

Many thanks for your reply; it is helpful.  I was indeed under the misunderstanding that bluetooth would be running under core 1, whereas you are suggesting that some of its tasks will be created on core 0

I tried the obvious switch: moving the measurement task to core 1 and running the BLE library on core 0:  the measurements now run without delays, but BLE fails to connect :-(
BLE does however advertise.  Note I removed the line to disable core 0 WDT

Is it possible that bludroid creates tasks (or interrupts) on both cores?

I will try to find out what tasks are running using VTaskList(()

Thanks again

Rob

chegewara, 2020-01-04:
Please check sdkconfig, or if you can menuconfig to see on which core bluetooh/bluedroid is running.

RobWilks, 2020-01-04:
I am running esp32-arduino.  I understand that esp32-elf is precompiled so
that changes to sdkconfig do not affect the code created by Arduino IDE.
Is that true?

Anyway the sdkconfig file would indicate that bluedroid is indeed pinned to
core 0, an excerpt follows.

Many thanks

Rob

---------------------------------------------
Excerpt from sdkconfig

#
# Bluetooth
#
CONFIG_BT_ENABLED=y

#
# Bluetooth controller
#
CONFIG_BTDM_CONTROLLER_MODE_BLE_ONLY=
CONFIG_BTDM_CONTROLLER_MODE_BR_EDR_ONLY=
CONFIG_BTDM_CONTROLLER_MODE_BTDM=y
CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN=3
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN=2
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN=0
CONFIG_BTDM_CONTROLLER_BR_EDR_SCO_DATA_PATH_HCI=
CONFIG_BTDM_CONTROLLER_BR_EDR_SCO_DATA_PATH_PCM=y
CONFIG_BTDM_CONTROLLER_BR_EDR_SCO_DATA_PATH_EFF=1
CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN_EFF=3
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN_EFF=2
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN_EFF=0
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_0=y
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_1=
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE=0
CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI=y
CONFIG_BTDM_CONTROLLER_HCI_MODE_UART_H4=

#
# MODEM SLEEP Options
#
CONFIG_BTDM_CONTROLLER_MODEM_SLEEP=y
CONFIG_BTDM_MODEM_SLEEP_MODE_ORIG=y
CONFIG_BTDM_MODEM_SLEEP_MODE_EVED=
CONFIG_BTDM_LPCLK_SEL_MAIN_XTAL=y
CONFIG_BLE_SCAN_DUPLICATE=y
CONFIG_SCAN_DUPLICATE_BY_DEVICE_ADDR=y
CONFIG_SCAN_DUPLICATE_BY_ADV_DATA=
CONFIG_SCAN_DUPLICATE_BY_ADV_DATA_AND_DEVICE_ADDR=
CONFIG_SCAN_DUPLICATE_TYPE=0
CONFIG_DUPLICATE_SCAN_CACHE_SIZE=20
CONFIG_BLE_MESH_SCAN_DUPLICATE_EN=
CONFIG_BLE_ADV_REPORT_FLOW_CONTROL_SUPPORTED=y
CONFIG_BLE_ADV_REPORT_FLOW_CONTROL_NUM=100
CONFIG_BLE_ADV_REPORT_DISCARD_THRSHOLD=20
CONFIG_BLUEDROID_ENABLED=y
CONFIG_BLUEDROID_PINNED_TO_CORE_0=y
CONFIG_BLUEDROID_PINNED_TO_CORE_1=
CONFIG_BLUEDROID_PINNED_TO_CORE=0
CONFIG_BTC_TASK_STACK_SIZE=8192
CONFIG_BLUEDROID_MEM_DEBUG=
CONFIG_CLASSIC_BT_ENABLED=y
CONFIG_A2DP_ENABLE=y
CONFIG_A2DP_SINK_TASK_STACK_SIZE=2048
CONFIG_A2DP_SOURCE_TASK_STACK_SIZE=2048
CONFIG_BT_SPP_ENABLED=y
CONFIG_HFP_ENABLE=y
CONFIG_HFP_CLIENT_ENABLE=y
CONFIG_HFP_AUDIO_DATA_PATH_PCM=y
CONFIG_HFP_AUDIO_DATA_PATH_HCI=
CONFIG_GATTS_ENABLE=y
CONFIG_GATTS_SEND_SERVICE_CHANGE_MANUAL=
CONFIG_GATTS_SEND_SERVICE_CHANGE_AUTO=y
CONFIG_GATTS_SEND_SERVICE_CHANGE_MODE=0
CONFIG_GATTC_ENABLE=y
CONFIG_GATTC_CACHE_NVS_FLASH=
CONFIG_BLE_SMP_ENABLE=y
CONFIG_SMP_SLAVE_CON_PARAMS_UPD_ENABLE=
CONFIG_BT_STACK_NO_LOG=y
CONFIG_BT_ACL_CONNECTIONS=4
CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST=y
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y
CONFIG_BLE_HOST_QUEUE_CONGESTION_CHECK=
CONFIG_SMP_ENABLE=y
CONFIG_BLE_ACTIVE_SCAN_REPORT_ADV_SCAN_RSP_INDIVIDUALLY=
CONFIG_BLE_ESTABLISH_LINK_CONNECTION_TIMEOUT=30
CONFIG_BT_RESERVE_DRAM=0xdb5c

chegewara, 2020-01-04:
Yes, thats correct.

RobWilks, 2020-01-04:
vTasklist() is not available on esp-idf
[https://github.com/espressif/arduino-esp32/issues/1089#issuecomment-364015566](url)
How can I check what runs where?
Rob

chegewara, 2020-01-04:
This is code i am using to check stack watermark, but you can find other info too:
```
static void tasksStatus()
{
    UBaseType_t uxArraySize = uxTaskGetNumberOfTasks();
    TaskStatus_t pxTaskStatusArray[uxArraySize];
    uint32_t pulTotalRunTime = 0;

    uxTaskGetSystemState(&pxTaskStatusArray[0], uxArraySize, &pulTotalRunTime);

    for(int i=0; i<uxArraySize; i++) {
        ESP_LOGI(__func__, "[%d]Task %s => stack remaining: %d", i, pxTaskStatusArray[i].pcTaskName, pxTaskStatusArray[i].usStackHighWaterMark);
    }
    heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
    ESP_LOGI(__func__, "stack remaining: %d", uxTaskGetStackHighWaterMark( NULL ));
}
```
Only thing its missing here is task affinity, but maybe someone else can help with it
@me-no-dev @lbernstone

RobWilks, 2020-01-04:
@chegewara 

Many thanks for your steers today. The original advice to pin BLE comms to core 0 was good. 

The only other changes needed were to:
- reduce the priority of the measurement task to 1
- increase priroity of the BLEcomms task to 2 and
- add a delay to the BLE loop so as to avoid triggering an error from task WDT

Code is here:
https://github.com/RobWilks/ESP32_BikeVibe

I'll now try out the stack watermark diagnostic - useful.  Thankyou again, Rob

RobWilks, 2020-01-10:
@chegewara 

I am using esp32-Arduino and compiling with Arduino IDE the tasksStatus() function you kindly shared earlier

At linking, I get the error:
C:\Users\rjwil\AppData\Local\Temp\arduino_build_558327\sketch\bikeVibeESP32CalFloatHRLogDeepSleep.ino.cpp.o:(.literal._Z16measureVibrationPv+0xb4): undefined reference to `uxTaskGetSystemState'
I assume the necessary function is not available on the Arduino pre-compile

Do you have any suggestions on how I proceed to get a log of the tasks running?

With many thanks,

Rob

chegewara, 2020-01-10:
Most likely it is not enabled in arduino.