nkolban/esp32-snippets #939: BLEDevice setPower having no affect on TxPower Level

gb53smith, 2020-01-14:
Using ESP32 BLE Arduino 1.0.1 in Platformio IDE
I have tried several power levels using this command immediately after BLEDevice::init
BLEDevice::setPower(ESP_PWR_LVL_P7);
The advertised TxPowerLevel is always 3 in the iPhone BLE Scanner App and the RSSI values
don't change. 
Seems like an old issue that was claimed to have been fixed?

gb53smith, 2020-01-31:
Fixed using source code from
https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE
This provides TxPower level 9.  Not sure if all statements are actually necessary.
```
/* The power level can be one of:
 * * ESP_PWR_LVL_N14
 * * ESP_PWR_LVL_N11
 * * ESP_PWR_LVL_N8
 * * ESP_PWR_LVL_N5
 * * ESP_PWR_LVL_N2
 * * ESP_PWR_LVL_P1
 * * ESP_PWR_LVL_P4
 * * ESP_PWR_LVL_P7
 */
  // Using maximum power for greatest range
  // Raises the TX Power from 3 to 9 as seen by the iPhone Client
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_DEFAULT);
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_CONN_HDL0);
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_CONN_HDL1);
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_CONN_HDL2);
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_CONN_HDL3);
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_CONN_HDL4);
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_CONN_HDL5);
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_CONN_HDL6);
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_CONN_HDL7);
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_CONN_HDL8);
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_ADV);
  BLEDevice::setPower(ESP_PWR_LVL_P7, ESP_BLE_PWR_TYPE_SCAN);
```

closed 2020-01-31 by gb53smith

kevin71246, 2021-03-10:
> Fixed using source code from
> https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE
> This provides TxPower level 9. Not sure if all statements are actually necessary.
>
> ```
> /* The power [...]

For the maximum range, "ESP_PWR_LVL_P9" should be used, which is the same as +9dBm.  See https://docs.espressif.com/projects/esp-idf/en/v4.1/api-reference/bluetooth/controller_vhci.html for more info.  I also have more details here: https://stackoverflow.com/a/66556561/3147100

gb53smith, 2021-03-10:
I think that ESP_PWR_LVL_P7 should also be equivalent to ESP_PWR_LVL_P9 for "backward compatibility" according to the document above.

MemeCornucopia, 2023-12-24:
no matter what i try its still -49 dbm

octotus, 2024-05-08:
I saw this discussion on Espressif forum - https://esp32.com/viewtopic.php?t=21011 -- looks like it is possible with ESP's API. Haven't tried it yet - but worth a try.