nkolban/esp32-snippets #940: BLE only for scan

Mawa94, 2020-01-16:
Hello, can you tell me some steps or, if you can, the library, where BLE will only scan for devices, not connect nor create no server or client.

I want it because full library is very heavy for only scanning purpouse.

Thanks, best regards.

chegewara, 2020-01-16:
Library itself is not very heavy, because its only wrapper, but it uses C++ libraries too. esp-idf library is pretty much heavy itself.
If you need only scan, then maybe (just maybe) its better to find examples that are using plain esp-idf.

Mawa94, 2020-01-16:
Im new on this, what is esp-idf? I think I know it but just in case I dont.

Mawa94, 2020-01-22:
I just edited this post because my mark is to make sure Bluetooth classic is disabled because I'm pretty sure I will not use nor scan any Bluetooth classic.