nkolban/esp32-snippets #942: Error token '{' when defining ble callbacks

peppeg85, 2020-01-26:
hello, i'm trying the ble functionality, mixing the documentation and the the basic example of arduino bluetooth server i have it work.
now i'm trying to modify it do set some characteristics callbacks to achieve the data sent from the 
smartphone, but i have the following error:

```
exit status 1
expected class-name before '{' token
```

about the row: 
`class MyCallback: public BLECharateristicCallback {`

what is this kind of error?

moreover, is my code corect? i assign two pointer to two characteristics id, and set the callbacks to the characteristics: 

```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"

#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID2 "9d2643bb-e4eb-4208-a93b-389161a9930a"

BLECharacteristic* pCharacteristic = NULL;

class MyCallback: public BLECharateristicCallback {
    void onRead(BLECharacteristic* pCharacteristic) {
      pCharacteristic->setValue("richiesta modifica");
  };
    void onWrite(BLECharacteristic* pCharacteristic) {
      Serial.println("rischiesta scrittura ",pCharacteristic);
  }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pCharacteristic->setCallbacks(new MyCallback());

   BLECharacteristic *pCharacteristic2 = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID2,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic2->setValue("Hello World says peppe");
  pCharacteristic2->setCallbacks(new MyCallback());
  
pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

is this the right way or i'm wrong?

thank you in advance

chegewara, 2020-01-26:
`BLECharateristicCallback` => `BLECharateristicCallbacks`

peppeg85, 2020-01-26:
Thank you,i Will try,buy Sorry,where should i find these API?thank you

peppeg85, 2020-01-26:
i have the same error:

```
more then a library for  "BLEDevice.h"

Usata: C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\BLE
exit status 1
expected class-name before '{' token
```

maybe i have an old ble library? but i didnt installed it

chegewara, 2020-01-26:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.h#L131