nkolban/esp32-snippets #943: Failure on BLE Connect

tristanmelton, 2020-01-28:
Hi!  Love the work you've done with BLE!  I'm having a bit of an issue getting my BLE Client to work.  For whatever reason, it seems like I'm getting stuck on the pClient->connect(*pAddress); call in my MyClient... I have enabled verbose debugging and have the following relevant log:

`
D (14502) BLEAdvertisedDevice: - setRSSI(): rssi: -88
D (14506) BLEAdvertisedDevice: Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 1a
D (14514) BLEAdvertisedDevice: Type: 0x07 (ESP_BLE_AD_TYPE_128SRV_CMPL), length: 16, data: 5c447fa4481b45ad2e40b66be0b46bbd
D (14525) BLEAdvertisedDevice: - addServiceUUID(): serviceUUID: bd6bb4e0-6bb6-402e-ad45-1b48a47f445c
D (14535) WatchOS: Advertised Device: Name: , Address: 41:98:5a:96:45:bc, serviceUUID: bd6bb4e0-6bb6-402e-ad45-1b48a47f445c
D (14550) BLEScan: >> stop()
V (14550) FreeRTOS: Semaphore giving: name: ScanEnd (0x3ffdafe0), owner: start
D (14556) BLEScan: << stop()
D (14560) WatchOS: Found our device!  address: 41:98:5a:96:45:bc
Found our device! address : 41:98:5a:96:45:bc
Made client
Client Initialized
D (14570) Task: >> runTask: taskName=Task
V (14572) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (14575) BLEDevice: >> createClient
V (14582) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 58:cf:9e:ba:9d:d2, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -91, ble_adv: ??, flag: 26 ([LE General Discoverable Mode] [Simultaneous LE and BR/EDR to Same Device Capable (Controller)] [Simultaneous LE and BR/EDR to Same Device Capable (Host)] ), num_resps: 1, adv_data_len: 17, scan_rsp_len: 0
I (14586) BLEDevice: add conn_id: 0, GATT role: client
V (14628) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
D (14629) FreeRTOS: Semaphore taking: name: RegEvt (0x3ffddfbc), owner: <N/A> for connect
V (14636) BLEUtils: [status: 0]
D (14644) FreeRTOS: Semaphore taken:  name: RegEvt (0x3ffddfbc), owner: connect
V (14572) FreeRTOS: << wait: Semaphore released: name: ScanEnd (0x3ffdafe0), owner: <N/A>
D (14657) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
V (14655) FreeRTOS: >> wait: Semaphore waiting: name: RegEvt (0x3ffddfbc), owner: connect for connect
V (14670) BLEUtils: GATT Event: ESP_GATTC_REG_EVT
V (14684) BLEUtils: [status: ESP_GATT_OK, app_id: 0x0]
D (14690) BLEClient: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
I (14697) gattClientEventHandler: register app id: 0, 0, gattc_if: 4
V (14704) FreeRTOS: Semaphore giving: name: RegEvt (0x3ffddfbc), owner: connect
V (14712) FreeRTOS: << wait: Semaphore released: name: RegEvt (0x3ffddfbc), owner: <N/A>
D (14720) BLEDevice: << createClient
Connecting... D (14723) BLEClient: >> connect(41:98:5a:96:45:bc)
D (14729) BLEClient: >> clearServices
D (14733) BLEClient: << clearServices
D (14737) FreeRTOS: Semaphore taking: name: OpenEvt (0x3ffde010), owner: <N/A> for connect
D (14745) FreeRTOS: Semaphore taken:  name: OpenEvt (0x3ffde010), owner: connect
V (14753) FreeRTOS: >> wait: Semaphore waiting: name: OpenEvt (0x3ffde010), owner: connect for connect
W (44764) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0100
W (44765) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0100
D (44772) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT
V (44778) BLEUtils: GATT Event: ESP_GATTC_DISCONNECT_EVT
V (44784) BLEUtils: [reason: ESP_GATT_CONN_CONN_CANCEL, conn_id: 0, remote_bda: 41:98:5a:96:45:bc]
D (44793) BLEClient: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT
E (44801) gattClientEventHandler: disconnect event, reason: 256, connId: 0, my connId: 255, my IF: 4, gattc_if: 4
V (44812) FreeRTOS: Semaphore giving: name: OpenEvt (0x3ffde010), owner: connect
V (44819) FreeRTOS: << wait: Semaphore released: name: OpenEvt (0x3ffde010), owner: <N/A>
D (44820) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
D (44827) BLEClient: << connect(), rc=0
V (44835) BLEUtils: GATT Event: ESP_GATTC_OPEN_EVT
Connected :)
V (44844) BLEUtils: [status: ESP_GATT_ERROR, conn_id: 0, remote_bda: 41:98:5a:96:45:bc, mtu: 0]
D (44854) BLEClient: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
D (44854) BLEClient: >> getService: uuid: bd6bb4e0-6bb6-402e-ad45-1b48a47f445c
V (44861) FreeRTOS: Semaphore giving: name: OpenEvt (0x3ffde010), owner: <N/A>
D (44869) BLEClient: >> getServices
D (44879) BLEClient: >> clearServices
D (44883) BLEClient: << clearServices
D (44887) FreeRTOS: Semaphore taking: name: SearchCmplEvt (0x3ffde118), owner: <N/A> for getServices
D (44896) FreeRTOS: Semaphore taken:  name: SearchCmplEvt (0x3ffde118), owner: getServices
V (44904) FreeRTOS: >> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffde118), owner: getServices for getServices

`

Seems to me that the semaphore is getting locked and never unlocked?  Not sure how to get around this.  My code can be found here: https://github.com/tristanmelton/OpenWatch/blob/master/WatchOS/main/hello_tft.cpp

gb53smith, 2020-01-29:
I have discovered the same problem.  I built a client server pair using Arduino 1.0.1 (last updated 2 years ago).  Everything works fine with the pair in the same room.    However, if the signal is weak then the client will hang.  ie will not return to the Ardunio loop().   It will find the advertised device but fail to connect.   I even tried ` if (pClient->connect(myDevice)) ` to skip the remaining code if the connection is unsuccessful.  I do get a  disconnect callback but after that I have to reset the Client with the Server in range.    The client must automatically recover from any interaction with the server.
I found a guide to creating my own updated Arduino library so that is my next move.

chegewara, 2020-01-29:
@tristanmelton Yes, it seems that in your case library stuck on semaphore in case you are trying to get services when connection was not successful.
It should not have happen, library should prevent it, but what you can do is to add:
`pClient->isConnected()` before you try to get services, or right after connect().
If returns true then proceed else return false.

gb53smith, 2020-01-29:
Thanks, I'll try the isConnected test.

I tried using the lastest source files from this repository.  I am using Platformio.  It reports version 0.4.3
Here is the response:
```
Attempting MQTT connection...
        MQTT Connected
Starting every 10 seconds...
Starting Scan after disconnect
Forming a connection to 24:6f:28:97:17:7a
 - Created client
[E][BLEClient.cpp:193] gattClientEventHandler(): disconnect event, reason: 62, connId: 0, my connId: 255, my IF: 4, gattc_if: 4
We are now connected to the BLE Server.
Now connected
Starting every 10 seconds...
Starting Scan after disconnect
Forming a connection to 24:6f:28:97:17:7a
 - Created client
[E][BLEClient.cpp:193] gattClientEventHandler(): disconnect event, reason: 62, connId: 0, my connId: 0, my IF: 4, gattc_if: 4
[E][BLEClient.cpp:193] gattClientEventHandler(): disconnect event, reason: 62, connId: 0, my connId: 255, my IF: 5, gattc_if: 5
Now connected
Starting every 10 seconds...
Starting Scan after disconnect
Forming a connection to 24:6f:28:97:17:7a
 - Created client
[E][BLEClient.cpp:193] gattClientEventHandler(): disconnect event, reason: 62, connId: 0, my connId: 0, my IF: 4, gattc_if: 4
[E][BLEClient.cpp:193] gattClientEventHandler(): disconnect event, reason: 62, connId: 0, my connId: 0, my IF: 5, gattc_if: 5
[E][BLEClient.cpp:193] gattClientEventHandler(): disconnect event, reason: 62, connId: 0, my connId: 255, my IF: 6, gattc_if: 6
Now connected
Starting every 10 seconds...
Starting Scan after disconnect
Forming a connection to 24:6f:28:97:17:7a
 - Created client
```
As you see I try restarting scan on disconnect but after 4 trys the ESP32 hangs and needs a reset with the server nearby.

chegewara, 2020-01-29:
I am preparing to build bigger project now, so today i wrote simple proof of concept app with platformio => arduino, esp-idf.
My simple app is performing ble scan publish over mqtt results for every found ble device.
So far i have to say it works great, very stable and fast. (i have just slightly modified arduino library)

From your logs i dont understand why you have 1, 2 and then 3 disconnect events, because after disconnected device should be removed from array.
My advice is to use library from arduino-esp32 repository and replace library you are using now.

gb53smith, 2020-01-29:
I thought the source files in this repository were more recent, with fewer bugs than the 2 year old ESP32 BLE Ardunio 1.0.1 library.   I don't get  error messages seen with 0.4.3 but the ESP32 still hangs after a few tries to connect as seen in this log file:
```
Starting Arduino BLE Client application...
Starting Scan at Setup
In Reconnect, WL_CONNECTED = 3
Connecting to GreenMountain   
.......
WiFi connected
IP address:
192.168.1.76
Attempting MQTT connection...
        MQTT Connected
Starting every 10 seconds...
Starting Scan after disconnect
Starting every 10 seconds...
Starting Scan after disconnect
Starting every 10 seconds...
Starting Scan after disconnect
Starting every 10 seconds...
Starting Scan after disconnect
Starting every 10 seconds...
Starting Scan after disconnect
Starting every 10 seconds...
Starting Scan after disconnect
Forming a connection to 24:6f:28:97:17:7a
 - Created client
onDisconnect
```
tried this as suggested:
```
pClient->connect(myDevice);
if  (pClient->isConnected())  //Skip the rest if this fails
{
  Serial.println(" - Connected to server");
...
```
And the ESP32 still hangs needing a reset and server in range.  There seems to be bug in the `connect` function.  I don't have an Expressif development environment to dive any deeper.  My client ESP32 also uses MQTT and WiFi to connect with HomeAssistant.  Everything is working fine except for this reliability issue with weak BLE signals.

chegewara, 2020-01-29:
Its `ESP32 BLE Ardunio 1.0.1` only because no one is updating version number.

Now your logs looks much better. What you can do is to try to change timeout value in this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/FreeRTOS.cpp#L76
May or may not help, but worth to try.

tristanmelton, 2020-01-29:
@chegewara Hm, interesting.  I've switched to using rote BLE calls, modifing the gatt_client example for now.  I have no problems connecting using that example.

gb53smith, 2020-01-29:
> What you can do is to try to change timeout value in this line:
> https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/FreeRTOS.cpp#L76
> May or may not help, but worth to try.

Found the parameter `portMAX_DELAY` so I search all the files in the src directory but could not find where it is set. Please enlighten me.  What value should I try? 

PS.
So someone is releasing new code and not changing the version number?   Isn't that important?

chegewara, 2020-01-29:
`portMAX_DELAY` == forever

Try 1000

gb53smith, 2020-01-29:
Brilliant!  I replaced `portMAX_DELAY` in `freeRTOS.h` and `freeRTOS.cpp ` with 1000.  It now works!
I can move my server in and out of radio range and the client recovers from disconnects without a hardware reset.

Seems like "forever' was a bad thing.  I don't understand how this Ardunio code compiles without error.  
Were some files missing in the Arduino build that define`portMAX_DELAY` ?  
How does this bug fix get merged into a new version?

chegewara, 2020-01-29:
`portMAX_DELAY` is defined in FreeRTOS, not in this library.

gb53smith, 2020-01-29:
I found that `portMAX_DELAY` is set to `0xffffffffUL` in
`C:\Users\XXX\.platformio\packages\framework-arduinoespressif32\
tools\sdk\include\freertos\freertos\portmacro.h `
When I eliminated the link to `X:\Arduino\libraries\ESP32_BLE_Arduino` Platformio found another library here: `\.platformio\packages\framework-arduinoespressif32\libraries\BLE` which must be more compatible with the FreeRTOS version used by Platformio because this works without any edits to library files.
For anyone else using Platformio here is my env:
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
```

chegewara, 2020-01-31:
`ESP32_BLE_Arduino` this is very old library and not updated for long time.

gb53smith, 2020-01-31:
A further update to help Platformio users.  I made a custom library called "BLEDevice" using source files from the url given in the library.json file I created in the Platformio project/lib/BLEDevice directory.
```
{
"name": "BLEDevice",
"version": "1.0.0",
"repository":
{
    "type": "git",
    "url": "https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE"
}
}
```
My application is working with a snapshot of these latest development files.

closed 2020-02-01 by tristanmelton