nkolban/esp32-snippets #945: BLE Notify MTU size

nvg-ict, 2020-02-02:
I hope this is the right place to answer this question:

Im experimenting with BLE for a project and I want to investigate the possibilities of transferring larger datafiles through BLE. ATM I have found an option where I was able to change the MTU size to a higher value (517) and read/write to such characteristic. 

Now I am trying to do the same for a characteristic which has the Notify property, however it cuts out after 20 bytes. Is this due to the fact that BLE Notify doesn't support a larger MTU size or am I overlooking something?

The only thing I have done so far in order to change the MTU size is call the following function: 
`  BLEDevice::setMTU(517); `

chegewara, 2020-02-02:
It is possible peer device you are connecting with is not accepting MTU higher than 23, or something goes wrong in your code.

nvg-ict, 2020-02-02:
Alright, thanks for the quick reply. Im guessing that there is some small detail I am overlooking. So far I have use the BLE_uart and BLE_server examples you guys have provided. I will keep digging around if you say it should be possible to increase the notify MTU size.

(I want to thank you guys for the dedication and support given on this library. So far this has been my best experience so far with nicely documented libraries and examples.)

nvg-ict, 2020-02-03:
I have looked around in the code and I saw this above the notify() function:

`/**
 * @brief Send a notify.
 * A notification is a transmission of up to the first 20 bytes of the characteristic value.  An notification
 * will not block; it is a fire and forget.
 * @return N/A.
 */
void BLECharacteristic::notify(bool is_notification) {
`

So if I want to send a larger packet with Notify I have to do this another way (for example presenting the large data packets on a different characteristic?).

This is a very simple program I have put together to test the larger payload (based on the uart example):
`/*
    Video: https://www.youtube.com/watch?v=oCMOYS71NIU
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleNotify.cpp
    Ported to Arduino ESP32 by Evandro Copercini

   Create a BLE server that, once we receive a connection, will send periodic notifications.
   The service advertises itself as: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
   Has a characteristic of: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E - used for receiving data with "WRITE" 
   Has a characteristic of: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E - used to send data with  "NOTIFY"

   The design of creating the BLE server is:
   1. Create a BLE Server
   2. Create a BLE Service
   3. Create a BLE Characteristic on the Service
   4. Create a BLE Descriptor on the characteristic
   5. Start the service.
   6. Start advertising.

   In this example rxValue is the data received (only accessible inside that function).
   And txValue is the data to be sent, in this example just a byte incremented every second. 
*/
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer *pServer = NULL;
BLECharacteristic * pTxCharacteristic;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint8_t txValue = 0;
std::string payload = "Dit is een lange string waarbij het einde een byte is dat veranderd kan worden voor een lange payload met verschil.";
int cnt = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();

      if (rxValue.length() > 0) {
        Serial.println("*********");
        Serial.print("Received Value: ");
        for (int i = 0; i < rxValue.length(); i++)
          Serial.print(rxValue[i]);

        Serial.println();
        Serial.println("*********");
      }
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::setMTU(517);
  BLEDevice::init("UART Service");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pTxCharacteristic = pService->createCharacteristic(
										CHARACTERISTIC_UUID_TX,
										BLECharacteristic::PROPERTY_NOTIFY
									);
                      
  //pTxCharacteristic->addDescriptor(new BLE2902());

  BLECharacteristic * pRxCharacteristic = pService->createCharacteristic(
											 CHARACTERISTIC_UUID_RX,
											BLECharacteristic::PROPERTY_WRITE
										);

  pRxCharacteristic->setCallbacks(new MyCallbacks());

  // Start the service
  pService->start();

  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  
  Serial.println("Waiting a client connection to notify...");
}

void loop() {

if (deviceConnected) {
        pTxCharacteristic->setValue(payload);
        pTxCharacteristic->notify();
        payload[119] = cnt + '0';
        cnt = cnt%10;
    delay(10); // bluetooth stack will go into congestion, if too many packets are sent
  }
    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("start advertising");
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
		// do stuff here on connecting
        oldDeviceConnected = deviceConnected;
    }
}`

Any help/tips/advice is welcome.

chegewara, 2020-02-03:
```
/**
 * @brief Send a notify.
 * A notification is a transmission of up to the first 20 bytes of the characteristic value.  An notification
 * will not block; it is a fire and forget.
 * @return N/A.
 */
```
This only very old description, when changing MTU was not implemented yet.
Generally max notification size that wont be truncated is MTU - 3 bytes.

Try setMTU after BLEDevice::init().

nvg-ict, 2020-02-04:
No difference, it still seems to cut short after 20 bytes. 

I have done an experiment combining a notify characteristic and a READ/WRITE characteristic. The READ/WRITE does show a large MTU size, the notify still cuts out at 20 bytes.

```void BLE_Facade::init()
{

  BLEDevice::init(DEVICE_NAME);
  BLEDevice::setMTU(517);

  pServer = BLEDevice::createServer();
  pService = pServer->createService(SERVICE_UUID);
  pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pNotifyCharacteristic = pService->createCharacteristic(
										CHARACTERISTIC_UUID_TX,
										BLECharacteristic::PROPERTY_NOTIFY
									);

  pNotifyCharacteristic->addDescriptor(new BLE2902());

  pCharacteristic->setCallbacks(new MyCallbacks());

  pServer->setCallbacks(this);

  pCharacteristic->setValue("Hello World says Nico, hopelijk kan ik nu een veel grotere dataload sturen maar dat hangt ervan af hoe dat nu met dat MTU werkt enzo.");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  DEBUG_COM_PRINTLN("Characteristic defined! Now you can read it in your phone!");
}

void BLE_Facade::deinit()
{
  BLEDevice::deinit();
}

void BLE_Facade::notify(const std::string& msg)
{
  pNotifyCharacteristic->setValue(msg);
  pNotifyCharacteristic->notify();
}
```

The BLE_Facade notify function gets called with a long string with a length of ~100 bytes but the NRFconnect app still only shows 20. Same for my own made Xamarin BLE app which only receives a payload of the 20 bytes. 

I'm using a Samsung S10 phone with Android 10 which if I recall correctly uses Bluetooth 5.0. Might it be an idea to test this on a true Bluetooth 4.2 device like a Samsung S7?

What puzzles me is that the MTU size change doesn't affect the NOTIFY property but it seems to change the READ/WRITE property.

chegewara, 2020-02-04:
When you test with smartphone try nRF connect app. After connecting to esp32 you will have 3 dots icon, next to server. You can change MTU in there and then see in logs if MTU is accepted, returned value in logs is real MTU, not necessary the same you requested.
When you change MTU, then you can turn on notifications and see results.

nvg-ict, 2020-02-05:
Ah you are my hero!

That was the trick. I was assuming that setting the MTU size was some sort of contract between the devices but it needs to be done both ways.

Thanks for helping out here, I now need to find a way to change the MTU size from the APP, but that should be doable. Thanks again for helping me out here.

closed 2020-02-05 by nvg-ict

gabor-one, 2020-09-22:
Hello Guys,
I do have the same issue. I would like to notify an app about a changed characteristic but it only sends the first 20 bytes. When reading MTU settings works as intended.

Does somebody found a way to notify with more than 20 bytes by default?

ashish-iottive, 2021-03-19:
> No difference, it still seems to cut short after 20 bytes.
> 
> I have done an experiment combining a notify characteristic and a READ/WRITE characteristic. The READ/WRITE does show a large MTU size, the notify still cuts out at 20 bytes.
> 
> ```
> {
> 
>   BLEDevice::init(DEVICE_NAME);
>   BLEDevice::setMTU(517);
> 
>   pServer = BLEDevice::createServer();
>   pService = pServer->createService(SERVICE_UUID);
>   pCharacteristic = pService->createCharacteristic(
>                                          CHARACTERISTIC_UUID,
>                                          BLECharacteristic::PROPERTY_READ |
>                                          BLECharacteristic::PROPERTY_WRITE
>                                        );
> 
>   pNotifyCharacteristic = pService->createCharacteristic(
> 										CHARACTERISTIC_UUID_TX,
> 										BLECharacteristic::PROPERTY_NOTIFY
> 									);
> 
>   pNotifyCharacteristic->addDescriptor(new BLE2902());
> 
> 
>   pCharacteristic->setCallbacks(new MyCallbacks());
> 
>   pServer->setCallbacks(this);
> 
>   pCharacteristic->setValue("Hello World says Nico, hopelijk kan ik nu een veel grotere dataload sturen maar dat hangt ervan af hoe dat nu met dat MTU werkt enzo.");
>   pService->start();
>   // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
>   BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
>   pAdvertising->addServiceUUID(SERVICE_UUID);
>   pAdvertising->setScanResponse(true);
>   pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
>   pAdvertising->setMinPreferred(0x12);
>   BLEDevice::startAdvertising();
>   DEBUG_COM_PRINTLN("Characteristic defined! Now you can read it in your phone!");
> }
> 
> void BLE_Facade::deinit()
> {
>   BLEDevice::deinit();
> }
> 
> void BLE_Facade::notify(const std::string& msg)
> {
>   pNotifyCharacteristic->setValue(msg);
>   pNotifyCharacteristic->notify();
> }
> ```
> 
> The BLE_Facade notify function gets called with a long string with a length of ~100 bytes but the NRFconnect app still only shows 20. Same for my own made Xamarin BLE app which only receives a payload of the 20 bytes.
> 
> I'm using a Samsung S10 phone with Android 10 which if I recall correctly uses Bluetooth 5.0. Might it be an idea to test this on a true Bluetooth 4.2 device like a Samsung S7?
> 
> What puzzles me is that the MTU size change doesn't affect the NOTIFY property but it seems to change the READ/WRITE property.

Could you provide the full code in the next comment

chegewara, 2021-03-19:
It is really not hard:
- MTU is something that is sort of "negotiated" and not all devices has to support it or size supported by device can be smaller,
- in that case used MTU is smallest supported by both peer devices,
- when you are testing with nRF connect you cave to explicitly set MTU in app after connecting, because app does not do it automatically, 
- READ/WRITE is not affected by MTU, because there is something like long read/write, which is sending many packets, when notify/indicate is sending only 1 packet.

ashish-iottive, 2021-03-22:
> It is really not hard:
>
> * MTU is something that is sort of "negotiated" and not all devices has to support it or size supported by device can be smaller,
> * in that case used MTU is smallest [...]

Thank You very much, I have found the way
Cheers!

patricioruizperret, 2021-04-02:
Hello.
Did you found how to do it?. Using _BLEDevice::setMTU(128);_ nothing change for me. The only way i found is to modify the MTU using the app nRF Connect, but i have to do it each time a turn on the ESP32...
Regards,
Patricio

chegewara, 2021-04-02:
> Hello.
> Did you found how to do it?. Using _BLEDevice::setMTU(128);_ nothing change for me. The only way i found is to modify the MTU using the app nRF Connect, but i have to do it each time a turn on the ESP32...
> Regards,
> Patricio

You have to do it with nRF connect each time, because that app does not do it automatically. You can do nothing with it.

patricioruizperret, 2021-04-03:
Ouch! That is not a good new!. I will search another way. Thanks!

joseitor, 2023-05-26:
Did you find another way to change MTU from ESP32?

patricioruizperret, 2023-05-28:
Hello, hola.
I could not modify the MRU size. Instead a sent a long message splitted in 2 or more messages, un to 20 char each. It worked fine

chegewara, 2023-05-29:
actually there is option  in nRF connect to change MTU on connect, just dig in config/options

joseitor, 2023-05-30:
> actually there is option in nRF connect to change MTU on connect, just dig in config/options

Yeah, I know, and doing that, I cand notify more than 20bytes.
But what I'm looking for, is a way to request that change from ESP32.

gabor-one, 2023-05-30:
Hey Jose,

The way MTU size negotiation works is: Server offers max MTU size and the
client can request MTU size change. If nRF connect cant change the MTU size
probably you are not setting Max MTU right in your code on the ESP.

gabor-one, 2023-05-30:
> The nRF connect CAN change MTU size.
> What I'm trying to do is ESP32 change as server MTU size without
> interaction from nRF connect. It's that possible?
>
> Before a used a BLE module (rn4870) that was able to send more than 20
> bytes as notification without change MTUsize in the app. I don't know how
> but it does, is the ESP32, a more powerful and modern chip unable to do the
> same?

> What I'm trying to do is ESP32 change as server MTU size without
interaction from nRF connect. It's that possible?
No, the client needs to request the change.

cyberFIRM, 2023-10-30:
Just as @chegewara mentioned MTU size is negotiated on connection time between client and sever. You can manually request the MTU size from the device via NRF connect but you will have to do it everytime when your device gets rebooted. Unfortunately, there is no API on gatt server side that supports this. 
There are 2 workarounds to set this in the firmware:

- If you are using IDF framework you will need to call esp_ble_gatt_set_local_mtu(uint16_t mtu) function before creating the server. But ofcourse it is only supported by BLUEDROID.
- If you are using NimBLE, you can call the ble_att_clt_tx_mtu(uint16_t conn_handle, uint16_t mtu) funtion in onConnect callback to change the MTU successfully. This function will exchange the MTU with the peer.