nkolban/esp32-snippets #946: Single server and multiple clients

kowshik1234, 2020-02-03:
Hey guys, 

I've been recently playing around with the ESP32 's BLE feature. So, as a part of it, I wanted to try out server-client configuration. Basically I will be having one server ESP32 server which connects to a client esp32 and my mobile too. When I send a string from mobile, it should reach the client esp32 through the server. here is a small picture of what I've been talking about. I hope this must be possible with ESP32's BLE peripheral.

[image]

chegewara, 2020-02-03:
Sure, it is possible.

kowshik1234, 2020-02-03:
Thanks, Can I have any example codes which I can develop on?

kowshik1234, 2020-02-05:
Any suggestions @chegewara

chegewara, 2020-02-05:
Use BLE client and BLE server as starting point and try to merge them.