nkolban/esp32-snippets #951: WriteCharacteristics doesn't work through ESP but is working through noble.

rahulmr-rently, 2020-02-08:
Hi All,
I am trying to work on a BLE lock which is currently being operated through a mobile app using the  react-native ble plx.  I was trying to make it work through the ESP32 so that it can be controlled remotely. I have prepared the data and is being sent to esp32 through aws iot. data is received but when trying to write to the lock it doesn't respond. 
In order to see whether it is something wrong with command or not I installed the noble in my mac and passed the same command through the noble write method and it works and lock responds.

chegewara, 2020-02-08:
Try with `uint8_t input[] = {...};`, unless you need to get length from String. But for testing purpose you can test it that way.
Maybe you write to wrong characteristic?

rahulmr-rently, 2020-02-09:
@chegewara  Thanks for the reply. I debugged and tested again with the BLE connect to check what comes from the ESP32 and from the noble.

In noble project ```

let buffer = encryptedData();// -> which returns a buffer
console.log(i + ": " + buffer.toString("hex")); //aa5510f0988fbef125a97eee5e7ae2beeaff79
```
The above aa5510f0988fbef125a97eee5e7ae2beeaff79 string value I pass and send it to aws iot to receive in the ESP and this is what is written to the ble device.

In the ble scanner app I could see the 

- HexValue-"6161353531306630393838666265663132356139376565653565376165326265656166663739"
- String value - "aa5510f0988fbef125a97eee5e7ae2beeaff79"

But noble takes the write() a buffer as input so characteristics.write(buffer) is received as the following in the ble scanner

- Hex Value- "aa5510f0988fbef125a97eee5e7ae2beeaff79"

- String Value- "No Value".

I find the aa5510f0988fbef125a97eee5e7ae2beeaff79 to reach as hex in the ble device when the esp32 characteristics.write(input) is called .The input is also string "aa5510f0988fbef125a97eee5e7ae2beeaff79"