nkolban/esp32-snippets #952: Increase BLE Beacon Scanning strength

Kevistar, 2020-02-09:
Hello, 

Previously, I opened up an issue to inquire about the methods to scan for beacon's UUID and it is working perfectly. However, while testing it I realize the RSSI picked up by the ESP32 module is lower compared to the RSSI measured by an android BLE beacon scanner app. (Example: At 1m, the beacon scanner app registered RSSI reading of -60 while the ESP32 module registered  RSSI reading of -75). Are there any methods to increase the scanning strength of the module itself? I would like to improve the scanning performance as it would be used in indoor positioning projects in the future. Here is the current code I am working on. 

`/*
Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEUUID.h>
#include <BLEAdvertisedDevice.h>
#include <BLEBeacon.h>
BLEScan* pBLEScan;
BLEBeacon id;
int scanTime = 10; //In seconds
#define CLEAR_RESULT           "00000000-0000-0000-0000-000000000000"

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks
{
   void onResult(BLEAdvertisedDevice advertisedDevice)
   {
        id.setData(advertisedDevice.getManufacturerData());   
        String identity = id.getProximityUUID().toString().c_str();

        if (identity == "74278bda-b644-4520-8f0c-720eaf059935")
       {
        //Print UUID 
        Serial.print("UUID :");
        Serial.println(id.getProximityUUID().toString().c_str());
        
        //Print RSSI
        Serial.print("RSSI :");
        Serial.println(advertisedDevice.getRSSI());   

        //Print Major
        Serial.print("Major :");
        Serial.println(id.getMajor());

        //Print Major
        Serial.print("Minor :");
        Serial.println(id.getMinor());
        Serial.println("");
       }
        id.setProximityUUID(BLEUUID(CLEAR_RESULT));   //Clear scan history
        id.setMajor(0);
        id.setMinor(0);
    }
};

void setup() {
 
    Serial.begin(115200);
    BLEDevice::init(""); 
    pBLEScan = BLEDevice::getScan(); //create new scan
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
    pBLEScan->setActiveScan(true);
    Serial.println("Scanning...");
   
}

void loop()
{
    BLEScanResults foundDevices = pBLEScan->start(scanTime);
    Serial.print("Devices found: ");
    Serial.println(foundDevices.getCount());
    Serial.println("Scan done!");
    pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
    delay(1000);
}`

chegewara, 2020-02-09:
Its matter of hardware, antenna, case etc.

Kevistar, 2020-02-09:
Could that mean my esp32 module is faulty and I should replace it with a new one?

chegewara, 2020-02-09:
Possibly, or any other factor is related.