nkolban/esp32-snippets #956: Crash on second call to BLEScan

Elmi77, 2020-02-26:
I make use of following code to scan for Bluetooth-devices:

```
   BLEDevice::init("");
   pBLEScan = BLEDevice::getScan(); //create new scan
   if (pBLEScan)
   {
      pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
      pBLEScan->setActiveScan(true); //set passive scan; active scan uses more power, but get results faster
      pBLEScan->setInterval(100);
      pBLEScan->setWindow(99);  // less or equal setInterval value
      BLEScanResults foundDevices = pBLEScan->start(5,false);
      pBLEScan->stop();
      pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
      delete pBLEScan;
      pBLEScan=0;
   }
   BLEDevice::deinit(true);

```
This code works properly on the very first call (called out of setup()) but when I try to call it the second time some minutes later (this time out of loop() but this does not seem to make a difference), it crashes with folowing error message:

`/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1443 (xQueueGenericReceive)- assert failed!
`

The whole Bluetooth-stuff is completely initialised and removed on every call due to memory problems - when I do not release this stuff, my application lacks enough memory.

Any idea what is wrong and how I can avoid this crash?

Thanks!

chegewara, 2020-02-26:
Remove this code:
`delete pBLEScan;`
As you can see in this code BLEScan is created only when m_pScan is null, but after deleting you have no option to set this variable to null and next time you have returned address of deleted BLEScan:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.cpp#L303

Elmi77, 2020-02-26:
OK, when I remove this line, the crash is gone - but from the second call to pBLEScan->start() no more bluetooth-devices are found, the callback does not return anything.

chegewara, 2020-02-26:
Thats possible, because you call deinit and all "static" classes like BLEScan or BLEAdvertising may be not valid any longer and may require reinit too.

Before library has been abandoned this option has not been tested (deinit and reinit), so it is possible it requires small changes in code to make it working properly.

For example add delete BLEScan in this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.cpp#L618
then set m_pScan = nullptr. The same for other "static" classes.

Elmi77, 2020-02-26:
What means the lib is abadoned? Isn't it maintained any more? What is the successor - which method of scanning BLE devices should be used instead?

Elmi77, 2020-02-26:
OK, found the answer to the question about the future of the lib - thanks!