nkolban/esp32-snippets #961: Can't get into void loop() after reading a characteristic value

SClp21, 2020-03-22:
Hello! 
I try to send the sensor's value read from one esp32 to another esp32. But the issue is that
after read a received sensor value it's show this error.
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: , Address: d0:03:df:a9:cd:bb, manufacturer data: 75004204010167d003dfa9cdbbd203dfa9cdba0b000000000000
BLE Advertised Device found: Name: , Address: 5e:5d:ce:13:01:8b, manufacturer data: 4c001005511c28273d
BLE Advertised Device found: Name: , Address: 44:5e:bd:87:c0:04, manufacturer data: 4c0010051118237bef, txPower: 12
BLE Advertised Device found: Name: , Address: 4b:83:34:ec:f0:c4, manufacturer data: 4c001005511c28273d
BLE Advertised Device found: Name: ESP32, Address: 30:ae:a4:9c:ff:26, serviceUUID: 383c25ab-2fb3-4018-97a6-3cec9824bcaf, txPower: 3
Yes, it contain the service
Forming a connection to 30:ae:a4:9c:ff:26
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
The characteristic value was: 1223.00 
Can notify?: YES
registering for notification
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d533f  PS      : 0x00060130  A0      : 0x800d1f94  A1      : 0x3ffc9600  
A2      : 0x00000000  A3      : 0x3ffc963e  A4      : 0x00000002  A5      : 0x00000000  
A6      : 0x3ffe1248  A7      : 0x3f019600  A8      : 0x800d4924  A9      : 0x3ffc95b0  
A10     : 0x3ffc95fc  A11     : 0x3ffe1248  A12     : 0x00000024  A13     : 0x227b01e7  
A14     : 0x3ffc95b0  A15     : 0x0000002d  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000030  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d533f:0x3ffc9600 0x400d1f91:0x3ffc9630 0x400d2045:0x3ffc9680 0x400d7b65:0x3ffc96f0 0x4008e089:0x3ffc9710

```
with the exception decoder
```
Decoding stack results
0x400d533f: std::__cxx11::basic_string  , std::allocator  >::_M_construct (char*, char*, std::forward_iterator_tag) at c:\users\automata\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/basic_string.tcc line 223
0x400d1f91: connectToServer() at C:\Users\Automata\Documents\Arduino\BLE_client/BLE_client.ino line 85
0x400d2045: loop() at C:\Users\Automata\Documents\Arduino\BLE_client/BLE_client.ino line 156
0x400d7b65: String::toInt() const at C:\Users\Automata\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\WString.cpp line 842
0x4008e089: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
### Sketch server
```
/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
    updates by chegewara
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <Wire.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"
#define SERVICE_UUID "383c25ab-2fb3-4018-97a6-3cec9824bcaf"
BLECharacteristic sensorSense (BLEUUID("beb5483e-36e1-4688-b7f5-ea07361b26a8"), BLECharacteristic::PROPERTY_READ
[... 215 lines cut ...]
    Serial.println("Total string is : " + totalString);
    String force = totalString.substring(0,5);
    //String alcohol = totalString.substring(6,11);
    /*Serial.print("Force : " + force);
    Serial.print(" | ");
    Serial.println("Alcohol rate : " + alcohol);
    if(alcohol.toInt()>1000){
      digitalWrite(Relay1, HIGH); // ส่งให้ไฟติด
      delay(1000); // ดีเลย์ 1000ms
    }else if(alcohol.toInt()<1000){
      digitalWrite(Relay1, LOW); // ส่งให้ไฟติด
      delay(1000); // ดีเลย์ 1000ms
    }*/
    String newValue = "Time since boot: " + String(millis()/1000);
    Serial.println("Setting new characteristic value to \"" + newValue + "\"");
    
    // Set the characteristic's value to be the array of bytes that is actually a string.
    pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
  }else if(doScan){
    BLEDevice::getScan()->start(0);  // this is just eample to start scan after disconnect, most likely there is better way to do it in arduino
  }
  
  delay(1000); // Delay a second between loops.
} // End of loop
```

chegewara, 2020-03-22:
I see its now commented out, but this backtrace points to String.toInt():
```
0x400d7b65: String::toInt() const at C:\Users\Automata\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\WString.cpp line 842
```
Maybe this is problem:
```
//String alcohol = totalString.substring(6,11);
    /*Serial.print("Force : " + force);
    Serial.print(" | ");
    Serial.println("Alcohol rate : " + alcohol);
    if(alcohol.toInt()>1000){
      digitalWrite(Relay1, HIGH); // ส่งให้ไฟติด
      delay(1000); // ดีเลย์ 1000ms
    }else if(alcohol.toInt()<1000){
      digitalWrite(Relay1, LOW); // ส่งให้ไฟติด
      delay(1000); // ดีเลย์ 1000ms
    }*/
```
In other hand crash is right after this log line:
```
Serial.println("registering for notification");
    
    //BLERemoteDescriptor *desc = writeChar->getDescriptor(BLEUUID((uint16_t)0x2902);
    //desc->writeValue(v,2);
    
    //pRemoteCharacteristic->registerForNotify(notifyCallback);
    pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)indicationOn,2);
    if(pRemoteCharacteristic->canNotify())
      Serial.println("Registering for Notification");
      pRemoteCharacteristic->registerForNotify(notifyCallback);
    Serial.println("Code reached here");
    connected = true;
```
Maybe i am missing something, but i dont see obvious problem with this code.
Try to comment line by line and see what is causing crash.

SClp21, 2020-03-22:
Thanks for reply!
 Yes, I commented the line after `pRemoteCharacteristic->registerForNotify(notifyCallback);`
as `Serial.println("Code reached here");` but this doesn't print out so I guess the problem is `pRemoteCharacteristic->registerForNotify(notifyCallback);`

chegewara, 2020-03-22:
I would suggest to add parenthesis:
```
if(pRemoteCharacteristic->canNotify()){
      Serial.println("Registering for Notification");
      pRemoteCharacteristic->registerForNotify(notifyCallback);
}
```

SClp21, 2020-03-26:
Thanks for your suggestion!
 Now my client can get into void loop() after I changed the descriptor in the server side to 0x2902

closed 2020-03-26 by SClp21