nkolban/esp32-snippets #962: Continues Scan using BLE_Scan example

marcoavaccaro, 2020-03-27:
Hello,

I'm using the example BLE_Scan to search and monitorate the state of various beacons. But the way the code works is not the best for my application.

The perfect way to scan for me is doing a continue scan and every time the same beacon is catch te code showns it again.

The example using the native sdk of esp32 works that way but i prefer to use the arduino IDE.

The image below shows how the esp32 native sdk works. It's possible to see that the esp read the same beacon in a window of time of 133ms (141ms in the other example).

[image]

Someone know or have a code to do continues scans?

marcoavaccaro, 2020-03-27:
I already set this option to True and still don't receive any duplicate reads.

[image]

Only different MACs are shown
[image]

chegewara, 2020-03-27:
Just tested and works.
Code:
```
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 0; //In seconds
BLEScan* pBLEScan;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), true);
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {
  // put your main code here, to run repeatedly:
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
  delay(2000);
}
```
Result:
```
18:39:36.147 -> Advertised Device: Name: , Address: ac:23:3f:a0:3a:fa, serviceUUID: 0000feaa-0000-1000-8000-00805f9b34fb 
18:39:36.212 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.212 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, manufacturer data: 4c00021550765cb7d9ea4e2199a4fa879613a492b29466ddce 
18:39:36.245 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, serviceUUID: cbbfe0e1-f7f3-4206-84e0-84cbb3d09dfc 
18:39:36.245 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.311 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.345 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, manufacturer data: 4c00021550765cb7d9ea4e2199a4fa879613a492b29466ddce 
18:39:36.345 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.378 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, manufacturer data: 2d0102000110f93fafcb98ce42ecbd00eebb2c50ef727286806f360d 
18:39:36.378 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, serviceUUID: cbbfe0e1-f7f3-4206-84e0-84cbb3d09dfc 
18:39:36.378 -> Advertised Device: Name: , Address: 58:20:25:6f:a2:2e, manufacturer data: 4c001005111ccf4311, txPower: 12 
18:39:36.411 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.477 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.477 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, manufacturer data: 2d0102000110f93fafcb98ce42ecbd00eebb2c50ef727286806f360d 
18:39:36.510 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, serviceUUID: cbbfe0e1-f7f3-4206-84e0-84cbb3d09dfc 
18:39:36.510 -> Advertised Device: Name: , Address: ac:23:3f:a0:3a:fa, serviceUUID: 0000feaa-0000-1000-8000-00805f9b34fb 
18:39:36.544 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.577 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.610 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.610 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, manufacturer data: 4c00021550765cb7d9ea4e2199a4fa879613a492b29466ddce 
18:39:36.643 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, manufacturer data: 2d0102000110f93fafcb98ce42ecbd00eebb2c50ef727286806f360d 
18:39:36.643 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.710 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.743 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.743 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, manufacturer data: 4c00021550765cb7d9ea4e2199a4fa879613a492b29466ddce 
18:39:36.776 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, manufacturer data: 2d0102000110f93fafcb98ce42ecbd00eebb2c50ef727286806f360d 
18:39:36.776 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, serviceUUID: cbbfe0e1-f7f3-4206-84e0-84cbb3d09dfc 
18:39:36.776 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
[... cut ...]
18:39:36.909 -> Advertised Device: Name: , Address: fc:01:7c:64:c1:20, serviceUUID: cbbfe0e1-f7f3-4206-84e0-84cbb3d09dfc 
18:39:36.909 -> Advertised Device: Name: , Address: 58:20:25:6f:a2:2e, manufacturer data: 4c001005111ccf4311, txPower: 12 
18:39:36.942 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
18:39:36.975 -> Advertised Device: Name: Le Gizmeaux (110), Address: d8:a0:1d:64:c1:6a, serviceUUID: f20c24a8-bde6-45e8-90f2-85bf180cedd2, txPower: 3 
```

marcoavaccaro, 2020-03-27:
Thank you! Worked perfectly.

I believe that changing the settings in .h archives when I started the code it rewrote the settings without setting duplicates.

Thank you again.

closed 2020-05-06 by marcoavaccaro