nkolban/esp32-snippets #964: Get Bluetooth device name

EvanKly, 2020-03-31:
Hello,

I started using Neil Kolblan Bluetooth library using his example BLE_SCAN (At the moment I didn't change anything). I am wondering if it's possible to get the name of bluetooth devices ? For now, that's what i've got in the serial monitor :

Scanning...
Advertised Device: Name: , Address: 7d:a4:76:30:04:f2, manufacturer data: 4c001006031a940c00a4, txPower: 12 
Devices found: 1
Scan done!

Thank you very much for your help. I'm not familiar with pretty much everything surrounding Bluetooth technology and security, sorry if it looks like a dumb question.

chegewara, 2020-03-31:
Hi,
lucky for me i am learning whole life and today i learned something very important. 
From advertising data i can say it is iPhone, and today i learned that iPhone does not advertise name if bluetooth app is running in background, like in this case:
```
 , Address: 7d:a4:76:30:04:f2, manufacturer data: 4c001006031a940c00a4, txPower: 12
```

EvanKly, 2020-03-31:
Good to know thanks !
I tried to disable bluetooth on my phone and this advertising data doesn't show anymore, so you're right it was my Iphone. I tried moving around with the ESP and found two other signals which are :

`Advertised Device: Name: , Address: 8c:85:90:35:dc:51, manufacturer data: 4c0010050b1c29d3eb `
`Advertised Device: Name: , Address: 79:21:61:c2:0d:a9, manufacturer data: 4c000c0e00639abc79ba2017665c4fcf8868 `

Same here, I can't get their names. Does that mean they are also Iphones ? If not, is there a way to get them ?

Thanks a lot ! I appreciate

chegewara, 2020-03-31:
Similar situation, 1st device is iPhone, 2nd device, i cant say for sure.

If you have android smartphone you can install nRF connect and use it to create and start advertising.

Leepq, 2020-06-29:
> If you have android smartphone you can install nRF connect and use it to create and start advertising.

If I do not use iPhone and I don't want to use nRF Connect, what other ways are there to make my Andriod device appear on the Serial Monitor?

Thanks.

chegewara, 2020-06-29:
Your own app on smartphone.

Leepq, 2020-06-30:
Are there ways to make your smartphone advertise in BLE other than nrfConnect? We must create our own app to force advertise BLE?
I am pretty sure nrfConnect advertising would allow the name of the smartphone to not appear as null.

chegewara, 2020-06-30:
nRF connect is advertising smartphone name. You just have to create advertising and add name.