nkolban/esp32-snippets #966: can you send integer data using BLE ?

adithrmwn24, 2020-04-05:
Hello,

I'm using ESP-Vroom-32 for my project. In this board, I connect soil moisture sensor V1.2 and Temperature Sensor DHT-11 Module. I cannot send integer value, but must be converted into a string first and then can be sent.

Is it possible to send integer data using BLE in ESP32 ?

chegewara, 2020-04-05:
Yes, you can send integer.
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteCharacteristic.h#L49
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteCharacteristic.h#L51

prakash-subash, 2020-05-21:
Dear chegewara, 

I need some more explanation about your comments.
I am also looking for sending integer data from server to client.

Thank you

chegewara, 2020-05-21:
Here is not too much to explain:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteCharacteristic.h#L51
sending integer value in range 0-255.

In other hand this can be used to send integer or float values of any kind:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteCharacteristic.h#L49

uint8_t is 1 byte long, for example int type is 4 bytes long. uint8_t* is pointer to first byte of any value, to make it more elegant it probably should be void*.
Now if you have `int val = 0` then `(uint8_t*) val` will point to first byte of val in memory, passing sizeof(int) as length will send all bytes of that integer.

prakash-subash, 2020-05-23:
Dear chegewara,

Thanks for you suggestion.

server:
syntax for sending integer 
pCharacteristic->setValue((uint8_t*)&line_current, 4);

client:
syntax for receiving integer
uint16_t value = pRemoteCharacteristic->readUInt16();

I have got exact data at client end what I expected from server.

Thanks 
Prakash S

chegewara, 2020-05-24:
> uint16_t 

This is sizeof 2 bytes, but you are sending 4 bytes. You have to keep consistent types when you are sending and reading data.