nkolban/esp32-snippets #967: getName() BLE ESP32 Library return alway NULL!

Pietro1994, 2020-04-08:
When I call the `getName()` function it returns the device name only of the very first scan, while in the subsequent NULL. Why?

this is the code:

`
void scanArea()
{
  Serial.print("\n#############################################################");
  Serial.print("\n###### My Address MAC BLE: ");
  Serial.print(MyMAC);
  Serial.print("\n#############################################################");
  pBLEScan = BLEDevice::getScan();
  pBLEScan->setActiveScan(true);  
  BLEScanResults results = pBLEScan->start(1); 
  Serial.println("\n-----------------------------Device--------------------------------");

  for (int i = 0; i < results.getCount(); i++) 
  {
    Serial.println("\n_______________________________________________________________________");
    BLEAdvertisedDevice device = results.getDevice(i);
    Serial.print("\nDevice Address ");
    Serial.print(i + 1);
    Serial.print(": ");
    MAC = device.getAddress().toString().c_str();
    Serial.print(MAC);
    Serial.print(" | Rssi: ");
    Rssi_device = device.getRSSI();
    Serial.print(Rssi_device);
    Name = device.getName().c_str();
    Serial.print(" | Nome Device: ");
    Serial.print(Name);
    Manufacturer = device.getManufacturerData().c_str();
    Serial.print(" | Manufacturer Device: ");
    Serial.print(Manufacturer);
  }
  Serial.print("\n__________________________________________________________________________");
}
`

what can I do to have the name of the device at each scan?

thanks.

chegewara, 2020-04-08:
Use this example as reference, especially this line:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_server/BLE_server.ino#L35

Pietro1994, 2020-04-09:
My ESP32 when scanning the area only finds the name in the first scan and the subsequent NULL. But thanks to your answer I managed to consider another parameter that was able to help me in my work. Thanks for your answer.
However the `getName()` has some problems, because I have seen that other people have the same problem.

closed 2020-04-09 by Pietro1994

chegewara, 2020-04-09:
Yes, i seen many similar issues opened and always it was users fault. I could always deliver logs that shows i can scan how many times i want with device name in result every single time.

Pietro1994, 2020-04-09:
Thanks for the time you have dedicated to me, because thanks to you I found another way that led me to the goal.
However the problem of `getName()` from things is due?
Why so many people have had this problem but nobody has managed to find a solution.

chegewara, 2020-04-09:
If you could use nRF connect as a tool to scan and see what data is advertised in peripheral that you want to scan with name, which esp32 dont want to scan with name then we could try to find why. When you find that peripheral is really advertising name you can copy raw data or paste screenshot and i will try to find reason.