nkolban/esp32-snippets #968: Why do I get a stack overflow in accept task ?

salocinx, 2020-04-11:
Hi kolban - thanks for sharing your C++ snippets! I am trying to get a TCP server up and running by doing the following:

```
SockServ ss(19881);
ss.start();
```

But I am getting a stack overflow in task `acceptTask`:

```
***ERROR*** A stack overflow in task acceptTask has been detected.
abort() was called at PC 0x40090720 on core 1

ELF file SHA256: eea8109c42aca3c154f033c3b49ca096a53c708d874a5d97a3fea90f01966cb5

Backtrace: 0x40090375:0x3ffc7f50 0x40090709:0x3ffc7f70 0x40090720:0x3ffc7f90 0x40093b63:0x3ffc7fb0 0x40092d88:0x3ffc7fd0 0x40092d3e:0x00000000 |<-CORRUPTED
```

I looked up the program counter on my current *.elf and got this information:

`PC 0x40090720: vApplicationStackOverflowHook + 20 in section .iram0.text`

Your `SockServ` snippet already assigns `8*1024` to the accept task. However, I tried to increase the stack size to to `16*1024`, to no avail; same error. Some idea what I am doing wrong?

chegewara, 2020-04-11:
Probably some changes in esp-idf that leads to this issue.
What you have to do is to increase stack size here:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/SockServ.cpp#L173

Maybe 10 * 1024 will be enough.

chegewara, 2020-04-11:
Sorry, i dont use this class, but i am guessing it has something to do with esp-idf version.
Code has been tested with v3.3 or older, and maybe some changes in esp-idf leads to it.

salocinx, 2020-04-14:
Hi. Thanks for your answer! Yes it seems like many things changed from the last 3.x release to 4.0. I have working C code for this purpose. I will try to form some C++ classes wrapping up the TCP und UDP procedures myself. Thank you!