nkolban/esp32-snippets #969: ESP32 Crashes when using UUID Filter

FlorianBre, 2020-04-14:
Hello Guys, 

I'm trying to establish a BLE connection from my ESP32 to another mikrocontroller.
This works perfectly fine but especially the following function call took around 11 seconds until the referece to the correct serice is found:

```ino

// Obtain a reference to the service we are after in the remote BLE server.
  pRemoteService = pClient->getService(serviceUUID);

```

To improve this I tried to use the UUID Filter in the BLEClient.cpp the following way:

```c++

std::map<std::string, BLERemoteService*>* BLEClient::getServices() {
/*
 * Design
 * ------
 * We invoke esp_ble_gattc_search_service.  This will request a list of the service exposed by the
 * peer BLE partner to be returned as events.  Each event will be an an instance of ESP_GATTC_SEARCH_RES_EVT
 * and will culminate with an ESP_GATTC_SEARCH_CMPL_EVT when all have been received.
 */
	log_v(">> getServices");
// TODO implement retrieving services from cache
	clearServices(); // Clear any services that may exist.
	static BLEUUID serviceTest("433be3cc-ee9e-d797-0a40-e86d902d99dc");
	esp_bt_uuid_t* testFilter = serviceTest.getNative();
	esp_err_t errRc = esp_ble_gattc_search_service(
		getGattcIf(),
		getConnId(),
		testFilter // Filter UUID
	);

	m_semaphoreSearchCmplEvt.take("getServices");
	if (errRc != ESP_OK) {
		log_e("esp_ble_gattc_search_service: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
		return &m_servicesMap;
	}
	// If sucessfull, remember that we now have services.
	m_haveServices = (m_semaphoreSearchCmplEvt.wait("getServices") == 0);
	log_v("<< getServices");
	return &m_servicesMap;
} // getServices
```
When using this Filter my EPS32 crashes. The debug log of this process looks like the following:

[V][BLEDevice.cpp:66] createClient(): << createClient
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEClient.cpp:97] connect(): >> connect(f8:f0:05:c6:25:85)
[D][BLEClient.cpp:460] handleGAPEvent(): BLEClient ... handling GAP event!
[I][BLEDevice.cpp:593] addPeerDevice(): add conn_id: 0, GATT role: client
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegEvt (0x3ffedbe8), owner: <N/A> for connect
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegEvt (0x3ffedbe8), owner: connect
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: RegEvt (0x3ffedbe8), owner: connect for connect
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEClient.cpp:159] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: RegEvt (0x3ffedbe8), owner: connect
[V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: RegEvt (0x3ffedbe8), owner: <N/A>
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: OpenEvt (0x3ffedd04), owner: <N/A> for connect
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: OpenEvt (0x3ffedd04), owner: connect
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: OpenEvt (0x3ffedd04), owner: connect for connect
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 40
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 40
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 40
[D][BLEClient.cpp:159] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:577] updatePeerDevice(): update conn_id: 4, GATT role: client
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[D][BLEClient.cpp:159] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: OpenEvt (0x3ffedd04), owner: connect
[V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: OpenEvt (0x3ffedd04), owner: <N/A>
[V][BLEClient.cpp:130] connect(): << connect(), rc=1
[V][BLEClient.cpp:378] getService(): >> getService: uuid: 433be3cc-ee9e-d797-0a40-e86d902d99dc
[V][BLEClient.cpp:414] getServices(): >> getServices
[V][BLEClient.cpp:72] clearServices(): >> clearServices
[V][BLEClient.cpp:79] clearServices(): << clearServices
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: SearchCmplEvt (0x3ffedd64), owner: <N/A> for getServices
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: SearchCmplEvt (0x3ffedd64), owner: getServices
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 20 0x14
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffedd64), owner: getServices for getServices
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEClient.cpp:460] handleGAPEvent(): BLEClient ... handling GAP event!
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 20 0x14
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEClient.cpp:460] handleGAPEvent(): BLEClient ... handling GAP event!
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 6
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 6
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 6
[D][BLEClient.cpp:159] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: SearchCmplEvt (0x3ffedd64), owner: getServices
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 18
[V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: SearchCmplEvt (0x3ffedd64), owner: <N/A>
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEClient.cpp:432] getServices(): << getServices
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 18
[V][BLEClient.cpp:395] getService(): << getService: not found
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d7192  PS      : 0x00060130  A0      : 0x800d24d9  A1      : 0x3ffd8700  
A2      : 0x00000000  A3      : 0x188a2451  A4      : 0x3ffd5348  A5      : 0xf3954d43  
A6      : 0xbb9c7783  A7      : 0x00010d38  A8      : 0x80081612  A9      : 0x3ffd8720  
A10     : 0x00004f59  A11     : 0x00000000  A12     : 0x00004f59  A13     : 0x000053c0  
A14     : 0x3ffd51ac  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000034  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff8  

Backtrace: 0x400d7192:0x3ffd8700 0x400d24d6:0x3ffd8760 0x400d2995:0x3ffd87b0 0x400db3b5:0x3ffd87d0 0x4008e455:0x3ffd87f0
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

Am I doing anything wrong using the UUID Filter? 
Thanks for your Advices in Advance.

chegewara, 2020-04-14:
Hi,
crash i most likely in next line, because you get nullptr and you are trying to use it.
This is relevant:
`getService(): << getService: not found`
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.cpp#L394

FlorianBre, 2020-04-14:
@chegewara Thanks for the response.

I am aware that the crash is caused by the nullptr :)
But I dont know why do I get the nullptr in return when using the UUID Filter.
If im not using the filter just the standard BLEClient.cpp file like this:

'''c
std::map<std::string, BLERemoteService*>* BLEClient::getServices() {
/*
 * Design
 * ------
 * We invoke esp_ble_gattc_search_service.  This will request a list of the service exposed by the
 * peer BLE partner to be returned as events.  Each event will be an an instance of ESP_GATTC_SEARCH_RES_EVT
 * and will culminate with an ESP_GATTC_SEARCH_CMPL_EVT when all have been received.
 */
	log_v(">> getServices");
// TODO implement retrieving services from cache
	clearServices(); // Clear any services that may exist.
	static BLEUUID serviceToc("433be3cc-ee9e-d797-0a40-e86d902d99dc");
	esp_bt_uuid_t* test = serviceToc.getNative();
	esp_err_t errRc = esp_ble_gattc_search_service(
		getGattcIf(),
		getConnId(),
		NULL
	);

	m_semaphoreSearchCmplEvt.take("getServices");
	if (errRc != ESP_OK) {
		log_e("esp_ble_gattc_search_service: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
		return &m_servicesMap;
	}
	// If sucessfull, remember that we now have services.
	m_haveServices = (m_semaphoreSearchCmplEvt.wait("getServices") == 0);
	log_v("<< getServices");
	return &m_servicesMap;
} // getServices
'''

The connection works and the log ist the following:

[V][BLEDevice.cpp:66] createClient(): << createClient
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEClient.cpp:97] connect(): >> connect(f8:f0:05:c6:25:85)
[D][BLEClient.cpp:461] handleGAPEvent(): BLEClient ... handling GAP event!
[I][BLEDevice.cpp:593] addPeerDevice(): add conn_id: 0, GATT role: client
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegEvt (0x3ffede78), owner: <N/A> for connect
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegEvt (0x3ffede78), owner: connect
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: RegEvt (0x3ffede78), owner: connect for connect
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEClient.cpp:159] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: RegEvt (0x3ffede78), owner: connect
[V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: RegEvt (0x3ffede78), owner: <N/A>
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: OpenEvt (0x3ffeded8), owner: <N/A> for connect
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: OpenEvt (0x3ffeded8), owner: connect
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: OpenEvt (0x3ffeded8), owner: connect for connect
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 40
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 40
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 40
[D][BLEClient.cpp:159] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:577] updatePeerDevice(): update conn_id: 4, GATT role: client
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[... 100 lines cut ...]
[V][BLERemoteCharacteristic.cpp:250] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: fc282607-9639-4566-9ab5-cabfb2400929
[D][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): Found a descriptor: Handle: 24, UUID: 00002902-0000-1000-8000-00805f9b34fb
[V][BLERemoteCharacteristic.cpp:294] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[V][BLERemoteCharacteristic.cpp:45] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 26, UUID: da0e685b-c2bd-4d1c-be3f-566cb00f3cc2
[V][BLERemoteCharacteristic.cpp:36] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 26 0x26, uuid: da0e685b-c2bd-4d1c-be3f-566cb00f3cc2
[V][BLERemoteCharacteristic.cpp:250] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: da0e685b-c2bd-4d1c-be3f-566cb00f3cc2
[D][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): Found a descriptor: Handle: 27, UUID: 00002902-0000-1000-8000-00805f9b34fb
[V][BLERemoteCharacteristic.cpp:294] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[V][BLERemoteCharacteristic.cpp:45] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 29, UUID: 30025770-8808-4830-bf6c-3a1de823fb69
[... cut ...]
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[V][BLERemoteCharacteristic.cpp:294] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[V][BLERemoteCharacteristic.cpp:45] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[V][BLERemoteService.cpp:209] retrieveCharacteristics(): << getCharacteristics()

But as I said without the filter it took me 11 seconds to get the reference pointer to the service. Maybe there is another option to reduce the time?

chegewara, 2020-04-14:
First of all it is very odd it takes so long to discover services. Please try to lower log level to info.

Now, please check if you get proper value in this call:
`esp_bt_uuid_t* testFilter = serviceTest.getNative();`
This should not be a problem, but depending on compiler 'NULL == nullptr' or 'NULL != nullptr`.
Now if testFilter is nullptr then getting filtered service may fail/return 0 services.
It is also small possibility this function is bugged when you pass service to filter.

FlorianBre, 2020-04-29:
Hello sorry that it took me so long to respond I had to get debugg hardware first.
I tested the function call:
```c
esp_bt_uuid_t* testFilter = serviceTest.getNative();
```
It does not return a nullptr but the proper UUID.
Do you know of somebody who used this Filter before?

Another Thing:
I tried to connect with the same BLE device with the ESP32 but reduced the numbers of services of this BLE device by one. 
The function call:
```c
// Obtain a reference to the service we are after in the remote BLE server.
  pRemoteService = pClient->getService(serviceUUID);
```

Took half the time than before (about 5.5 seconds). Does the amount of services offered by the server BLE device influence the time that much?

chegewara, 2020-04-29:
Its still very strange it takes so long to discover services. Could you try with nRF connect to see if android will discover all services faster.
Just one thing, android is caching services, so it may be very quick if services are already cached, but you can force nRF connect to re-discover services in top right corner.

FlorianBre, 2020-04-30:
I've already tried that. It takes significantly less time. About 2-3 seconds.