nkolban/esp32-snippets #970: Connect to multiple BLE Servers

kasperkls02, 2020-04-16:
Hello. 
Looking to connect to multiple BLE Servers, bike HR and Power Meter. 
I can succesfully connect to each device individually, but cannot figure out how to have to connections going on simultaniusly. I cannot do round robin as i need to get updates every second for each sensor.

Any one having example code connecting to multiple server?

Regards Kasper

h2zero, 2020-05-05:
@kasperkls02 you need to create a BLEClient instance for each connection, connect to the first with instance 1, scan, then connect to the second with instance 2 etc..

NbrTDB, 2020-11-19:
@h2zero I've tried this myself with a couple of servers (one `BLEClient` each), and while it seems perfectly content to connect to both of them, subsequently trying to disconnect from either one disconnects both at the same time. 

Is that expected behaviour? I would've thought using `BLEClient->disconnect()` would only have disconnected that specific server.

h2zero, 2020-11-19:
@NbrTDB That is not the expected behavior. I have not seen this in my testing, I will try to reproduce that result when I get home.

NbrTDB, 2020-11-19:
@h2zero Thanks for the quick response! I don't suppose you could let me know what modifications need to be made when you have a moment? I have been poking around the library but must confess I haven't figured out what needs to change. Any suggestions would be much appreciated.

h2zero, 2020-11-19:
I don't have access to my files at the moment and honestly I'm not sure anymore as to the specific code I changed as it's been a couple years since I did it. I use NimBLE now and I have ported this library to use it and fully supports multiple connections. 

Unless you have a specific reason to use bluedroid I'd suggest trying my NimBLE repo. 
You can check it out [here](https://github.com/h2zero/esp-nimble-cpp) or [here (arduino)](https://github.com/h2zero/NimBLE-Arduino) if you're interested. 

I'll have a look to see what code changes need to made to this repo anyway when I get home.

h2zero, 2020-12-18:
Found the modifications, there are many in this file but I think the only one needed to fix this is to change all the instances of 
```
if(m_gattc_if != gattc_if)
    break;
```
to:
```
if(m_gattc_if != gattc_if)
    return;
```
in the `BLEClient::gattClientEventHandler`