nkolban/esp32-snippets #974: Arduino Example for BLE ASync scan

DeeJayMX, 2020-05-01:
Hi everyone, everythings is in the title.

I'm looking for make an ASync scan non-blocking as it is explained on the other issue but I can't do it in arduino.

I'm a real beginer so please be comprehensive.

I'm using this to start my scan and

`  BLEDevice::init("");
  BLEDevice::setMTU(20);  
  pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), true);
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(5, EndofScan);
  pBLEScan->clearResults();
`

and i'm parsing only the call back so I force clearing at the start of the callback

but the scan is always blocking

Thanks for everything, my ESP is working with the library but not at it best I think.

DeeJayMX, 2020-05-01:
I think my answer is here but I can't make it work:
`bool           start(uint32_t duration, void (*scanCompleteCB)(BLEScanResults), bool is_continue = false);`

chegewara, 2020-05-01:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_scan/BLE_scan.ino

DeeJayMX, 2020-05-01:
The example you gave is blocking the "flow", as I explained it.

DeeJayMX, 2020-05-01:
I was talking about an arduino example for this:
https://github.com/nkolban/esp32-snippets/issues/496

chegewara, 2020-05-01:
> I think my answer is here but I can't make it work:
> `bool start(uint32_t duration, void (*scanCompleteCB)(BLEScanResults), bool is_continue = false);`

Ok, so why you cant make it to work? Maybe i can help.

DeeJayMX, 2020-05-02:
Because when I'm trying the function start() with more parameters, the IDE warn me that I'm not putting the good parameters for the function.

I want to create a scan in background, with a function at the end of the scan like in the "C" style example.
I don't find an example with the ASync function mode of Scan.

Like I said I'm not a real developper, I started with Blink on Arduino...

So with an example it's very easier for me to understand and make good use of the library.
Thanks in advance

DeeJayMX, 2020-05-02:
An example in arduino style..

h2zero, 2020-05-05:
@DeeJayMX the code you posted will not block since you passed a callback function, if you did not pass that function it would then use the blocking method.

One other comment, remove `BLEDevice::setMTU(20);` or change the value to 23 or more as that is the BLE spec default.

DeeJayMX, 2020-05-05:
The callback is working, but the scan keep the flow blocked till the end of the scan.
With this code I have to wait for callback or wait for the end of scan.

I want, like in the C example, to use the scan in background and with callback :
No blocking code and callbacks when something is found and when it's finish.

I'm using OLED device and if I put some OLED code in the callback, the display don't work...

As I said I'm a noob, I don't have the capacity to rewrite the lib.. So if a Jedi master can put me on the way could be really great.

chegewara, 2020-05-06:
Long time ago there was request to implement non blocking scan function and it is implemented, but i dont know why it is not working for you, i never been using it.
The reason i never been using it is i like to use regular freertos tasks. I am creating task that can run only ble sacn for example and i can control it in that task. You can for example stop scan after some time (run timed scan) then clear results to free used heap and start ir again, or you can check how much free heap left and if its small then stop scan, free scan results and start again.
There is plenty options how to use it and you have main loop to do whatever you want/need.

h2zero, 2020-05-06:
>I'm using OLED device and if I put some OLED code in the callback, the display don't work...

I would suggest not doing this, keep the callback code minimal, save the data and process it in main loop or create a task as @chegewara mentioned. If I'm interpreting your code correctly, the scan will not block. Also there is no need for `pBLEScan->clearResults();` after `pBLEScan->start()` as the next time you call it this way it will do that for you. 

Pseudo code example:
```

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
        if(advertisedDevice == deviceWanted) {
            data = advertisedDeviceData;
            updateDisplay = true;
        }
    }
};

void updateDisplayData(datatype data) {
    /*display code here*/
}

void EndofScan() {
    pBLEScan->start(5, EndofScan);
}

void setup() {
    BLEDevice::init(""); 
    pBLEScan = BLEDevice::getScan(); 
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), true);
    pBLEScan->setInterval(1349);
    pBLEScan->setWindow(449); 
    pBLEScan->setActiveScan(true); 
    pBLEScan->start(5, EndofScan);
}

void loop() {
    if(updateDisplay) {
        updateDisplayData(data);
        updateDisplay = false;
    }
}
```
Something like that should work at least. Without seeing your code there's not much more I can suggest.

h2zero, 2020-05-06:
> Long time ago there was request to implement non blocking scan function and it is implemented, but i dont know why it is not working for you, i never been using it.

@chegewara This is kind of funny, I've never used the blocking method 😄. Many ways to skin a cat, as the saying goes.

DeeJayMX, 2020-05-06:
Thanks I will test it soon,
I tried that way before and it didn't work...
I may have miss something.

@chegewara you said that you can put the scan in a RTOS task, so no problem to use that way ?

chegewara, 2020-05-06:
> @chegewara you said that you can put the scan in a RTOS task, so no problem to use that way ?

Not at all, i am using it most of the time this way.

radaromega, 2023-12-06:
For anyone wondering, what @h2zero suggested here: https://github.com/nkolban/esp32-snippets/issues/974#issuecomment-624410991 works nicely. Thanks! It will perform the scan (asynchronous) and then call `EndofScan` on completion which then calls the `scan` method again keeping it running until some exiting/stopping condition.

nevets963, 2023-12-06:
For anyone wondering, what @h2zero suggested here: https://github.com/nkolban/esp32-snippets/issues/974#issuecomment-624410991 works nicely. Thanks! It will perform the scan (asynchronous) and then call `EndofScan` on completion which then calls the `scan` method again keeping it running until some exiting/stopping condition.