nkolban/esp32-snippets #976: random question about the new esp idf in platformio?  library, component,  or include?

wegunterjr, 2020-05-05:
I have this in my lib folder right now and have been using it that way for a while...
PLatformio added the cmake build tools to allow the esp idf to be built more like normal, but now I am not sure what do with the snippets...   are y'all including it as a lib or in the component folder, or are you putting it in the include folder?  
Either way, what are your working ways to use the cmakelists.txt?

chegewara, 2020-05-05:
Hi,
this library is more or less prepared to be used with cmake, but not sure how it will behave with platformio and cmake they build.
I put a lot time and effort to help with this library, but recently @h2zero started to porting nimble to arduino library with almost 100% compatible library. Its worth to mention, because nimble is ble only and this will make it more portable in heap and flash usage.
Really worth to try.

wegunterjr, 2020-05-06:
ah..because in arduino there is so much issue with the memory size of the libraries and not able to run BLE and WiFi at the same time?  I can do it with esp-idf, but couldn't with Arduino when I first started this journey...

chegewara, 2020-05-06:
https://github.com/h2zero/NimBLE-Arduino

h2zero, 2020-05-06:
Thanks for the mention @chegewara 😃.

@wegunterjr Yes NimBLE is a significant improvement in resource usage. If you are using IDF 3.3+ and would like to try it I would suggest using the IDF version of the library https://github.com/h2zero/esp-nimble-cpp. NimBLE-Arduino isn't really meant for IDF use right now, since IDF has NimBLE already it just needs the cpp wrapper in `project/components`.

wegunterjr, 2020-05-06:
Thanks for that. I am updating to IDF 4 at this time, so working through those changes now.
I was able to get it included by adding this to my CMakeLists at the top level:

`set(EXTRA_COMPONENT_DIRS "components/cpp_utils")`

and then in the cpp_utils folder, I added this:

```
# Edit following two lines to set component requirements (see docs)
set(COMPONENT_REQUIRES
  "console"
  "fatfs"
  "json"
  "mdns"
  "nvs_flash"
)
set(COMPONENT_PRIV_REQUIRES )

file(GLOB COMPONENT_SRCS
  LIST_DIRECTORIES false
  "*.h"
  "*.cpp"
  "*.c"
  "*.S"
)
set(COMPONENT_ADD_INCLUDEDIRS ".")

register_component()
```

My question now, is, if I have a 2nd Library next to cpp_utils in the components folder, how does it know about cpp_utils.  I tried adding it in the COMPONENT_REQUIRES , but that doesn't seem to work...

thoughts?

chegewara, 2020-05-07:
It should works, but i would suggest to ask this question on  forum. Angus will answer it best.