nkolban/esp32-snippets #977: Problem using u8g2, ST7920 and esp32 using the ESP-IDF

shaharsery, 2020-05-11:
Hi, I'm trying to create a simple project with the u8g2 library, an ST7920 LCD and the ESP32 using eclipse and the ESP-IDF but I can't get the LCD to show anything.
I've created a simple project using this [guide](http://www.lucadentella.it/en/2017/10/30/esp32-25-display-oled-con-u8g2/)
When I connect the same setup and use the Arduino IDE with the 'HelloWorld' example and
U8G2_ST7920_128X64_F_SW_SPI everything works fine.

This is my main app:

#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include <string.h>
#include <u8g2.h>

#include "sdkconfig.h"
#include "u8g2_esp32_hal.h"
#include "u8g2_esp32_hal.c"

#define PIN_CLK 18
#define PIN_MOSI 23
#define PIN_RESET 22
#define PIN_CS 5

void app_main() {
	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.clk   = PIN_CLK;
	u8g2_esp32_hal.mosi  = PIN_MOSI;
	u8g2_esp32_hal.cs    = PIN_CS;
	u8g2_esp32_hal.reset = PIN_RESET;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_t u8g2; // a structure which will contain all the data for one display
	u8g2_Setup_st7920_s_128x64_f(
		&u8g2,
		U8G2_R0,
		u8g2_esp32_spi_byte_cb,
		u8g2_esp32_gpio_and_delay_cb);  // init u8g2 structure

	u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,

	u8g2_SetPowerSave(&u8g2, 0); // wake up display
	u8g2_ClearBuffer(&u8g2);
	u8g2_SetFont(&u8g2, u8g2_font_ncenB08_tr);
	u8g2_DrawStr(&u8g2, 0,15,"Hello World!");
	u8g2_SendBuffer(&u8g2);

	vTaskDelete(NULL);
}

I'm pretty new to all of this so I might be missing something pretty simple, any help would be much appreciated,
Thanks!

PratikGupta114, 2022-11-14:
@nkolban I have been facing the same problem, kindly help.

lspaula, 2023-09-23:
Same issue here. I am using the same config. It compiles but nothing is shown on the display.

LucasDFranchi, 2023-09-25:
According to this [reply](https://www.esp32.com/viewtopic.php?t=16967) on the ESP forum, it is necessary to configure the default value of the chip selector pin. This pin is initially set by default when using the Arduino Framework.

So, it's necessary to change this [line ](https://github.com/tuupola/esp_examples/blob/8a935d15b56aaff12719d329e6039cb07c9e5b98/003-u8g2-ssd1306/main/u8g2_esp32_hal.c#L68C36-L68C36) from '0' to SPI_DEVICE_POSITIVE_CS.

``` C
spi_device_interface_config_t dev_config;
dev_config.address_bits     = 0;
dev_config.command_bits     = 0;
dev_config.dummy_bits       = 0;
dev_config.mode             = 0;
dev_config.duty_cycle_pos   = 0;
dev_config.cs_ena_posttrans = 0;
dev_config.cs_ena_pretrans  = 0;
dev_config.clock_speed_hz   = 10000;
dev_config.spics_io_num     = u8g2_esp32_hal.cs;
dev_config.flags            = SPI_DEVICE_POSITIVE_CS;
dev_config.queue_size       = 200;
dev_config.pre_cb           = NULL;
dev_config.post_cb          = NULL;
//ESP_LOGI(TAG, "... Adding device bus.");
ESP_ERROR_CHECK(spi_bus_add_device(HSPI_HOST, &dev_config, &handle_spi));
```