nkolban/esp32-snippets #978: How do you add a pairing code/passkey to a bluetooth service?

hzhh110, 2020-05-18:
`
void initBLE(const char*name){
        BLEDevice::init(name); 
	BLEServer *pServer = BLEDevice::createServer();
	pServer->setCallbacks(new MyServerCallbacks());
	BLEService *pService = pServer->createService(SERVICE_UUID);
  pTXCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_TX, BLECharacteristic::PROPERTY_NOTIFY);
	pTXCharacteristic->addDescriptor(new BLE2902());
	BLECharacteristic *pRXCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_RX, BLECharacteristic::PROPERTY_WRITE);
	pRXCharacteristic->setCallbacks(new MyCallbacks());
	pService->start();
	pServer->getAdvertising()->start();
}
`

chegewara, 2020-05-18:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/StaticPIN/StaticPIN.ino

This is one possibility.

hzhh110, 2020-05-21:
> https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/StaticPIN/StaticPIN.ino
> 
> This is one possibility.

No corresponding function was found!
BLESecurity *pSecurity = new BLESecurity();
pSecurity->setStaticPIN(123456); (No found setStaticPin)

chegewara, 2020-05-21:
Then maybe you have outdated library. 
Where you get library from?
How do you use it, with esp-idf or arduino ide or platformio?

hzhh110, 2020-05-23:
> Then maybe you have outdated library.
> Where you get library from?
> How do you use it, with esp-idf or arduino ide or platformio?

for platformio

chegewara, 2020-05-23:
Strange, it should works with platformio.

hzhh110, 2020-06-09:
> Strange, it should works with platformio.
> 
> Also depends which library is linked. Because it may be this library, which was not updated for long time:
> https://github.com/nkolban/ESP32_BLE_Arduino
> 
> My suggestion to to copy arduino BLE library to lib folder in project:
> https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE

https://github.com/nkolban/ESP32_BLE_Arduino and  https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE    BLESecurity  no have find  setStaticPIN???

chegewara, 2020-06-09:
Yes, today i realized that arduino version does not have staticPIN implemented. Its enough to replace BLESecurity files with thse from this library.