nkolban/esp32-snippets #981: How to connect a BLE server with bonding password

ramenia, 2020-06-06:
Hi,
I am trying to use ESP32 as a client and my server device has a bonding password. I couldnt find any example for how to connect a device which has bonding password. Is there any example ? Thank you.

chegewara, 2020-06-06:
https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/tests/BLETests/Arduino/security

ramenia, 2020-06-07:
No! If there is a password for bonding a device, these examples can not connect to that device. BLE Security callbacks never calls. Logs stops at connected to server.

ramenia, 2020-06-07:
I am sure something is wrong. Think like this there is a server which has a pairing passkey. How can i connect to that server with esp32. It is not an other esp32 it is a nrf52832