nkolban/esp32-snippets #983: Using notifyCallback function to write to OLED doesnt work

Fixatron, 2020-06-10:
I'm trying to make a similar project to the one here
https://github.com/CuriousTimo/ESP32/blob/master/ESP32_OLED_HRM/ESP32_OLED_HRM.ino
https://www.youtube.com/watch?v=iCKIIMrphtg

I want to use the Notification Callback function to display.clearDisplay();display.display(); but the display buffer never clears. 
What I need to do is to use loop, and I only want to write to the screen when I get a notification.

chegewara, 2020-06-10:
> but the display buffer never clears.

This question is to display driver author

Fixatron, 2020-06-11:
Cool, for the mean time I am running a timed loop that updates the display from values grabbed from the notifyCallback() function and stored as a String stringName;