nkolban/esp32-snippets #987: MacOS Big Sur dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found

SwiftySquirrel, 2020-06-29:
`Traceback (most recent call last):
  File "esptool.py", line 57, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports.py", line 29, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_posix.py", line 31, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_osx.py", line 32, in <module>
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
Failed to execute script esptool`

Worked before the macOS update, can’t even compile an empty code file with nothing but
`#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>

void setup() {
}

void loop() {
}
`

chegewara, 2020-06-29:
And how it is related to this library?
It is not even used in your code, even if you have includes.

closed 2020-06-29 by chegewara

hpwit, 2020-07-19:
I have just moved to Bug Sur beta and same issue :(