nkolban/esp32-snippets #988: About the Manufacturer Data scanned.

Leepq, 2020-06-30:
So I have seen [this](https://github.com/nkolban/esp32-snippets/issues/964#issuecomment-606589174) issue which got me thinking... Does a manufacturer data that starts with 4c001 belong to Apple always? I have done some testing and found that manufacturer data that starts with ff033 is associated with my Samsung device. Where do I go to see this information about Manufacturer Data? If I used the ff033, on the Bluetooth website for company identifiers, do I search 0x0033? How does the Hexadecimals link?

chegewara, 2020-06-30:
https://stackoverflow.com/questions/23626871/list-of-company-ids-for-manufacturer-specific-data-in-ble-advertising-packets

Leepq, 2020-06-30:
Thanks, however, when I searched for Samsung the Hexadecimals show 0x02DE and 0x0075 but I got this...
manufacturer data: ff03656661 which I assume as 0x00ff? But that shows Typo Products, LLC? I am confused by this still...

chegewara, 2020-06-30:
its more like 0x03ff

Leepq, 2020-06-30:
That would mean Withings?

Leepq, 2020-06-30:
But my phone is a Samsung, how did it become a smart scale name? This is weird

chegewara, 2020-06-30:
I cant answer that. Even if your smartphone is samsung that does not means it cant advertise and mimic any company. You can create advertising with apple id and mimic iBeacons.

Leepq, 2020-06-30:
Ah, thanks. I thought there was a way to differentiate by Manufacturer ID but it seems impossible now.

closed 2020-06-30 by Leepq

chegewara, 2020-06-30:
Usually it is, when device is sending true manufacturer data, if any.