nkolban/esp32-snippets #990: BLEAdvertisedDevice getPayload incorrect data

nickb512, 2020-07-07:
It might be operator error, not sure, but I have been looking at for long enough I can't see the wood for the trees now.

I'm trying to read the temperature from a BLE beacon but the payload I'm getting from my sketch differs from what nRF Connect app shows and what the beacon documentation suggests it should be.

The raw payload in nRF is 0x020106020A000A16ABFE70BF0A0122017C but in my sketch I get 0x3412BAAB2C000000010000001052FF3F00

The docs of the beacon I'm looking at are here and its Section 13 T&H package: http://doc.mokotechnology.com/index.php?s=/page/144#13.%20BeaconX%20Pro-T&H

The relavent section from the sketch is:

```
      BLEAdvertisedDevice advertisedDevice = foundDevices.getDevice(i);
      BLEAddress address = advertisedDevice.getAddress();
      uint8_t* payloadRaw = advertisedDevice.getPayload();
      size_t payloadLength = advertisedDevice.getPayloadLength();
      
      SerialMon.printf("BLE:    Result %i [%s] payload: 0x", i, address.toString().c_str());
      for (int i=0; i<payloadLength; i++) {
        if (payloadRaw[i] < 16) SerialMon.print("0");
        SerialMon.print(payloadRaw[i], HEX);
      }
      SerialMon.println();
```
It is inside a loop over the scan results

chegewara, 2020-07-07:
Hi,
it not possible you get 2 such different packets with this library from 1 device. Its from other device or this device is advertising 2 packets that are having different meanings, i have few devices around me that are doing this. Like TVs.

nickb512, 2020-07-07:
I'm not trying to get 2 different packets. The beacon is configured to just advertise the single frame and appears to be doing so (nRF Connect shows the correct data coming from the beacon, as per the documentation) but the ESP32 prints out totally different data for the same device

chegewara, 2020-07-07:
Me and many others are using it and we can confirm it works. So, which version are you using, is it installed from here or from arduino?

Dont get me wrong, im not saying this library is bugs free, is not, but that part is working, we tested it last time with eddystone beacons.

nickb512, 2020-07-07:
I'm also not saying my code is right either... but I have spent a fair bit of time on it and I don't seem to be getting any closer.

Do you have an example of a known working sample of code that scans and prints the payloads of all found devices to the console I can compare to?

I'm using the version installed in Arduino IDE as part of the ESP32 Board Manager version 1.0.4 which seems to suggest it is up to date?

chegewara, 2020-07-07:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_scan/BLE_scan.ino
You can add your code to `onResult` callback.

nickb512, 2020-07-07:
I'll give that a go tomorrow. As the expert on this subject (I bow to your knowledge) does my code above look about right to you?

chegewara, 2020-07-07:
Yes, i dont see anything wrong in that code. I would use `SerialMon.printf("%02x", payloadRaw[i])` to debug hex values, but only because i have mixed feelings about arduino and i dont know this API very good.

Benni77, 2022-10-17:
I'm pretty sure it's not working correctly. I have the exact same issue here.
What I'm using is the `scanCompleteCallback`. What in my opinion goes wrong is this pointer: 	`uint8_t* m_payload;`.
On ScanComplete, there is what ever it is on the address of the former payload.

Can this be possible? I tried to save the pointer on the `BLEAdvertisedDevice::parseAdvertisement` method, which worked perfectly!

markdegrootnl, 2022-12-02:
Can confirm
Payload as a result of `pBLEScan->start` is different from payload in `pAdvertisedDeviceCallbacks`.