nkolban/esp32-snippets #997: notify truncating to 20 bytes beside MTU is set.

racsnet, 2020-07-30:
Hello all
I try to send 36 Bytes as Server to a BLE connected Device.
But when i call notify i get the message, that my Payload is truncated.

src:

    #define BLE_MTU 517
    ...
    void setup() {
      BLEDevice::init(TAG);
      BLEDevice::setMTU(BLE_MTU);`
    ...
    void loop() {
    ...
      ESP_LOGI(TAG, "localMTU: %d", BLEDevice::getMTU());
      pRTxCharacteristic->notify();

LOG (relevant):

    [I][main.cpp:192] loop(): localMTU: 517
    ...
    [W][BLECharacteristic.cpp:513] notify(): - Truncating to 20 bytes (maximum notify size)

LOG (all):

    rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    flash read err, 1000
    ets_main.c 371 
    ets Jun  8 2016 00:22:57

    rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    configsip: 0, SPIWP:0xee
    clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
    mode:DIO, clock div:2
    load:0x3fff0018,len:4
    load:0x3fff001c,len:1044
    load:0x40078000,len:8896
    load:0x40080400,len:5828
    entry 0x400806ac
    [I][main.cpp:115] setup(): setMTU
    [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegisterAppEvt (0x3ffdcd08), owner: <N/A> for registerApp
    [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegisterAppEvt (0x3ffdcd08), owner: registerApp
    [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
    [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffdcf60), owner: <N/A> for createService
    [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffdcf60), owner: createService
    [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffdd1c4), owner: <N/A> for executeCreate
    [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffdd1c4), owner: executeCreate
    [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
    [D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=0000ffe1-0000-1000-8000-00805f9b34fb to service: UUID: 0000ffe0-0000-1000-8000-00805f9b34fb, handle: 0x0028
    [D][BLECharacteristic.cpp:90] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: 0000ffe1-0000-1000-8000-00805f9b34fb, service: UUID: 0000ffe0-0000-1000-8000-00805f9b34fb, handle: 0x0028
    [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffdd64c), owner: <N/A> for executeCreate
    [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffdd64c), owner: executeCreate
    [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
    [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffddc4c), owner: <N/A> for executeCreate
    [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffddc4c), owner: executeCreate
    [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
    [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: StartEvt (0x3ffdd284), owner: <N/A> for start
    [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: StartEvt (0x3ffdd284), owner: start
    [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
    [I][BLEDevice.cpp:554] getAdvertising(): create advertising
    [D][BLEDevice.cpp:556] getAdvertising(): get advertising
    [D][BLEAdvertising.cpp:196] start(): - no services advertised
    [D][BLEDevice.cpp:556] getAdvertising(): get advertising
    [D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 0]
    [D][BLEDevice.cpp:556] getAdvertising(): get advertising
    [D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 1]
    [D][BLEDevice.cpp:556] getAdvertising(): get advertising
    [D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 6]
    [D][BLEDevice.cpp:556] getAdvertising(): get advertising
    [D][BLEAdvertising.cpp:196] start(): - no services advertised
    [I][main.cpp:146] setup(): localMTU: 517
    [D][BLEDevice.cpp:556] getAdvertising(): get advertising
    [D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 0]
    [D][BLEDevice.cpp:556] getAdvertising(): get advertising
    [D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 1]
    [D][BLEDevice.cpp:556] getAdvertising(): get advertising
    [D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 6]
    [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
    [D][BLEServer.cpp:356] onConnect(): BLEServerCallbacks
    [D][BLEServer.cpp:357] onConnect(): BLEServerCallbacks
    [D][BLEServer.cpp:358] onConnect(): BLEServerCallbacks
    [D][BLEDevice.cpp:556] getAdvertising(): get advertising
    [D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 20]
    [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
    [D][BLEDevice.cpp:556] getAdvertising(): get advertising
    [D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 20]
    [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
    [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: SetValue (0x3ffddb30), owner: <N/A> for <Unknown>
    [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: SetValue (0x3ffddb30), owner: <Unknown>
    [D][BLECharacteristic.cpp:285] handleGATTServerEvent():  - Response to write event: New value: handle: 2a, uuid: 0000ffe1-0000-1000-8000-00805f9b34fb
    [D][BLECharacteristic.cpp:288] handleGATTServerEvent():  - Data: length: 4, data: 66110077
    [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unkno[w][FreeRTOS.cpp:189] take(): Semaphore taking: name: SetValue (0x3ffddb30), owner: <N/A> for <Unknown>
    [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: SetValue (0x3ffddb30), owner: <Unknown>
    [I][main.cpp:192] loop(): localMTU: 517
    [D][BLECharacteristic.cpp:782] onNotify(): BLECharacteristicCallbacks
    [D][BLECharacteristic.cpp:783] onNotify(): BLECharacteristicCallbacks
    [W][BLECharacteristic.cpp:513] notify(): - Truncating to 20 bytes (maximum notify size)
    nknownECharacter[Dtic.cpp:794] onStat[B():eviEeharp:t02isticCSerbacEventHa[BlECh)racattistvercppntHa] oeSt [esp_g BLEifara]te.isticCallbacks

any help on this?

chegewara, 2020-08-01:
`[I][main.cpp:192] loop(): localMTU: 517`
It is just local MTU, peer device you are connectint with has to send request to "negotiate" MTU.
Without it esp32 does know what message size peer device can receive and handle, so it has to use default 23.

racsnet, 2020-08-02:
Thanks for your reply. Sadly i have not control over the other Side. It is a Bikecontroler with a HM-10 Module. Is there a way to overwrite the MTU?

chegewara, 2020-08-02:
No.

What are you trying to do? Such devices usually have protocol how to interact with it.

racsnet, 2020-08-02:
It is an Ebike (Rayvolt Cruzer) i want to build an BikeComputer using a ESP32 and two Displays. For Bikes from Rayvolt there is an APP in Android. I was able to decompile this APP and looking into it. What I found is, that there are two types of Messages the Bike exchanges with the APP:
Status (Speed, Battery Level, such stuff) and Params (Configuration of the Bike). First I build a DummyBike to let the APP think there is a Bike. This works so long. But the APP requests the Params and expects 36 Bytes. 
The Code in the APP looks similiar to:

`if( (ble.data_available()) && (ble.data_length == 36) ) { parseParams() }`

chegewara, 2020-08-02:
There is 2 things you are mixing.

If app is requesting more data than MTU it is ok, because it will be sent in 2 or more packets and concatenated, but it is not the same if you want to send notify.

Like i said, there is protocol and you just show it is. Protocol is to read data from device and expect exactly 36 bytes.

racsnet, 2020-08-02:
Thank you very much for this explaination. Im ok with sending the data packet by packet. As long as ble.data_available() triggers only if all packets are transmitted.
Do you have a hind on how to do this?
here is, what I do atm:

    if (send_params) {
      pRTxCharacteristic->setValue("Hallo BLE World ich hab Probleme, mit langen Daten");
      ESP_LOGI("RayvoltDummy", "localMTU: %d", BLEDevice::getMTU());
      pRTxCharacteristic->notify();
      send_params = false;
    }

chegewara, 2020-08-03:
Obviously you are missing something in protocol.
Maybe there is some option to write to characteristic or set some value that will tell HC-10 to update MTU. Or maybe there is no option to send notifications. Did you check is that device is trying to write to CCC descriptor? It is not mandatory, but it is some clue.

You may need to do more reverse engineering, but there is no way to send more than 20 bytes if HC-10 is not requesting MTU. You may try to use wireshark.

racsnet, 2020-08-03:
Nothing in the Source of the APP about the MTU.
I will investigate it more with wireshark. But this needs a litle time because i first have to grab an Android Device wich is able to log the BLE connection. (thanks to Huawei ... they simply ignore "Bluetooth-HCI-Snoop-Protocoll Option")

Thank you very much for your help on this problem.

racsnet, 2020-08-04:
Problem solved.
The APP simply accepts multiple packets. I haved overlooked a part in the code, where bytes are stored for later use if length is not enought. 

Thank you again @chegewara

closed 2020-08-04 by racsnet