nkolban/esp32-snippets #999: Try to read characteristic value but nothing Show up

greatwuu, 2020-08-03:
Hi, I'm a newbie to ble. I try to read a value from the characteristic UUID like battery level, device name and current time. By far 
device name can be read easily without any problem but other goes south with current time and battery level the output from serial monitor look like those value can be read but nothing shows up. But I can read those values when use another phone with nRF Connect to read it, Did I need to pair with passcode before read those values??. Below is my code modified from the BLE_client example.

`/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * author unknown
 * updated by chegewara
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("0000180f-0000-1000-8000-00805f9b34fb");
static BLEUUID serviceUUID2("00001800-0000-1000-8000-00805f9b34fb");
static BLEUUID serviceUUID3("00001805-0000-1000-8000-00805f9b34fb");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("00002a19-0000-1000-8000-00805f9b34fb");
static BLEUUID    charUUID2("00002a00-0000-1000-8000-00805f9b34fb");
static BLEUUID    charUUID3("00002a2b-0000-1000-8000-00805f9b34fb");
String My_BLE_Address = "62:57:3a:f2:8c:37";

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("data: ");
    Serial.println((char*)pData);
}

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
  }

  void onDisconnect(BLEClient* pclient) {
    connected = false;
    Serial.println("onDisconnect");
  }
};

bool connectToServer() {
    Serial.print("Forming a connection to ");
    Serial.println(myDevice->getAddress().toString().c_str());
    
    BLEClient*  pClient  = BLEDevice::createClient();
    Serial.println(" - Created client");

    pClient->setClientCallbacks(new MyClientCallback());

    // Connect to the remove BLE Server.
    pClient->connect(myDevice);  // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
    Serial.println(" - Connected to server");

    // Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println(" - Found our service1 : Battery Service");

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println(" - Found our characteristic1 : Battery Service");
    // Read the value of the characteristic.
    if(pRemoteCharacteristic->canRead()) {
      std::string value = pRemoteCharacteristic->readValue();
      Serial.print("The characteristic value was: ");
      Serial.println(value.c_str());
    }
    
    // Obtain a reference to the service we are after in the remote BLE server.
    pRemoteService = pClient->getService(serviceUUID2);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println(" - Found our service2 : Device Name");

// Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID2);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println(" - Found our characteristic2 : Device Name");

    // Read the value of the characteristic.
    if(pRemoteCharacteristic->canRead()) {
      std::string value = pRemoteCharacteristic->readValue();
      Serial.print("The characteristic value was: ");
      Serial.println(value.c_str());
    }

        // Obtain a reference to the service we are after in the remote BLE server.
    pRemoteService = pClient->getService(serviceUUID3);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println(" - Found our service3 : Current Time");

// Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID3);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println(" - Found our characteristic3 : Current Time");

    // Read the value of the characteristic.
    if(pRemoteCharacteristic->canRead()) {
      std::string value = pRemoteCharacteristic->readValue();
      Serial.print("The characteristic value was: ");
      Serial.println(value.c_str());
    }

    if(pRemoteCharacteristic->canNotify())
      pRemoteCharacteristic->registerForNotify(notifyCallback);

    connected = true;
    return true;
}
/**
 * Scan for BLE servers and find the first one that advertises the service we are looking for.
 */
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
 /**
   * Called for each advertising BLE server.
   */
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());

    // We have found a device, let us now see if it contains the service we are looking for.
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)) {

      BLEDevice::getScan()->stop();
      myDevice = new BLEAdvertisedDevice(advertisedDevice);
      doConnect = true;
      doScan = true;

    } // Found our server
  } // onResult
}; // MyAdvertisedDeviceCallbacks

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");

  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 5 seconds.
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(5, false);
} // End of setup.

// This is the Arduino main loop function.
void loop() {

  // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are 
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    if (connectToServer()) {
      Serial.println("We are now connected to the BLE Server.");
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  }
  delay(1000); // Delay a second between loops.
} // End of loop`

[image]

chegewara, 2020-08-03:
You are trying to print string, but the value is most likely not string:
`Serial.println(value.c_str());`

greatwuu, 2020-08-04:
Oh! Thank you, I'll try to find ways to prints the hex value

greatwuu, 2020-08-04:
> You are trying to print string, but the value is most likely not string:
> `Serial.println(value.c_str());`

Or this library is on development for reading hex/binary from characteristic I do some research on these issues repo and found this [https://github.com/nkolban/esp32-snippets/issues/596#issue-339059560](url) . And I still don't know how to print those values out. By far I've read and research for readValue() function it seems that it already read values into string type and store by this command.  `std::string value = pRemoteCharacteristic->readValue();` 

How I can read hex or binary data from characteristics? Or I need to research for something more.

Thank you.

chegewara, 2020-08-04:
Try to print read value length to see if there is any value. 
You can print bytes with `Serial.write` or use `ESP_LOG_BUFFER_HEX`.

greatwuu, 2020-08-04:
I try to add other command to print byte but the result is still the same.
[image]

I add this to my sketch.
`if (pRemoteCharacteristic->canRead()) {
    std::string value = pRemoteCharacteristic->readValue();
    int j = value.length();
    Serial.print("Data length = ");
    Serial.println(j);
    Serial.print("The characteristic value was: ");
    //Serial.println(value.c_str());
    for (int i = 0; i < value.length(); i++)
    {
      Serial.write(value[i]);
    }
    Serial.println("");
  }`

And when I try to add `ESP_LOG_BUFFER_HEX("value", value, value.length());` it can't compile or maybe I use the wrong buffer pointer. If the problems is a pointer please give me some advice I still confuse with pointer

greatwuu, 2020-08-04:
I see some discussion on [https://github.com/nkolban/esp32-snippets/issues/666](url)

but still didn't understand that notifyCallback function where does it get input from I did not see this function use any input.

this function declare with 
`static void notifyCallback(BLERemoteCharacteristic* pBLERemoteCharacteristic,uint8_t* pData,size_t length,bool isNotify)`

chegewara, 2020-08-04:
here you can use that notation, because you are giving length:
`ESP_LOG_BUFFER_HEX("value", value.c_str(), value.length());`
so it does not matter if there is 0x0 in middle, or any non printable value, or you can do it that way:
`ESP_LOG_BUFFER_HEX("value", (uint8_t*)value.c_str(), value.length());`
It makes no difference here, because IIRC buffer is type of void*

greatwuu, 2020-08-04:
> here you can use that notation, because you are giving length:
> `ESP_LOG_BUFFER_HEX("value", value.c_str(), value.length());`
> so it does not matter if there is 0x0 in middle, or any non printable [...]

it can compile but still no data coming out did I need to pair or bond before reading Battery data because I've check the value.legth() result is = 0 so I have no idea.

chegewara, 2020-08-04:
> Anyway I use nRF Connect to emulate BLE service from my mobile phone.

Did you set any value in nRF for that characteristic?

greatwuu, 2020-08-04:
[image]

Haha, I didn't see any setting did it should have. When hit the add service button the application only let me choose a service but have no rights to set value for it.

chegewara, 2020-08-04:
For sure there is option to set value, at least on android it is.

greatwuu, 2020-08-05:
Update I've added basic security to my sketch now I have output in string form 
[image]

Next step is to convert this value to hex.

closed 2020-08-05 by greatwuu