nkolban/esp32-snippets #1000: What is the default bitrate on ESP 32 BLE Arduino?

haorfani, 2020-08-04:
Hi all,

I am using a simple multiserver - client example on ESP32 WROOM board. I am trying to find out what is the default bitrate and if I can change it in this library?

Best,
Haris

chegewara, 2020-08-06:
There is no such thing like default bitrate. You can try to send as fast as you can until you get congest event, which is also not implemented here, but you can listen to it with custom callbacks.

haorfani, 2020-08-06:
I did not really understand, how can I find out the current bitrate I am using?

chegewara, 2020-08-06:
Bitrate of what? There is no such thing in ble.

haorfani, 2020-08-06:
If you want to calculate how many bits you are sending per second in packet transfer? How do you do this?

chegewara, 2020-08-06:
"manually". I am sending in a lopp as fast as i can knowing how many bytes i am sending or receiving by the seconds esp32.

I could achieve about 100kbitps IIRC. There is topic on forum i answered some time ago.

haorfani, 2020-08-11:
So BLE 4.2 has a hardware specification of 1 Mbps as bitrate, I guess you cannot achieve this bitrate with BLEDevice library, right?

chegewara, 2020-08-12:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_throughput/throughput_client
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_throughput/throughput_server

This library should have comparable bitrate.

haorfani, 2020-08-12:
OK thanks!

closed 2020-08-12 by haorfani

chegewara, 2020-08-12:
And here is my old post:
https://www.esp32.com/viewtopic.php?f=2&t=2221&start=10#p59431

It is with OTA over BLE and bitrate is over 250kbitps, but it may be android app limitation or lack of optimization.