nkolban/esp32-snippets #1010: [question] onWrite callback doesn't get called in the BLE_write example

Suyash458, 2020-09-02:
I'm using an ESP32 Dev Kit and the `BLE_write` example code from [here](https://github.com/nkolban/ESP32_BLE_Arduino/tree/master/examples/BLE_write). While I'm able to connect to the ESP and read its custom characteristic using the BLE Scanner Android app, values that are written to the characteristic don't turn up on the Serial monitor. Am I missing something here?

chegewara, 2020-09-06:
It depends how and what you are trying to print values to serial logs.

Suyash458, 2020-09-07:
@chegewara I haven't modified the example (following the steps [here](https://randomnerdtutorials.com/esp32-bluetooth-low-energy-ble-arduino-ide/)) and I'm writing simple ASCII text to the characteristic.

Suyash458, 2020-10-13:
The following example works, closing this issue for now
https://gist.github.com/Suyash458/fcf5c7b93fe7139460515bb7a1a5641d

closed 2020-10-13 by Suyash458