nkolban/esp32-snippets #1011: ESP32 not discoverable under CSR8510 Bluetooth adapter

xelons, 2020-09-07:
Hi,

I can scan and find ESP32 on my iPhone as well as on some Android devices, however, it is not shown when using a Bluetooth USB adapter (CSR8510). The Bluetooth adapter is a standard module, it can scan all my other BLE enabled devices except for the ESP32.

Does anyone know why this is the case?
My advertising code looks like this:
`BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->addServiceUUID(BLEUUID(SERVICE_UUID));
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);
  pAdvertising->setScanFilter(false,false);  // functions that help with iPhone connections issue
   pServer->startAdvertising();`

chegewara, 2020-09-07:
Did you try with nRF connect app on mobile?

xelons, 2020-09-07:
@chegewara yes I tried the nRF app, does not show there either. I'm starting to suspect that my Bluetooth USB dongle needs a driver update to be able to start discovering. I'm currently downloading the CSR Bluetooth stack driver to replace the standard windows driver and will see how it goes.

chegewara, 2020-09-07:
If nRF connect app does not see esp32 then its app or hardware issue (esp32).

xelons, 2020-09-07:
I use the same esp32 on my iPhone and it connects with the app I developer, I can discovered it without issue, it is also discoverable on iPad and my Samsung Android phone.

It is not discoverable however on the CSR8510 Bluetooth adapter, please see attached picture:
https://i.imgur.com/9N9U9iA.png

xelons, 2020-09-08:
It is also not discoverable on macos :
https://i.imgur.com/0PABEkV.png

chegewara, 2020-09-08:
I still think its hardware issue, maybe very weak signal, i dont see other explanation.