nkolban/esp32-snippets #1013: Is it possible to see when other device is scanning for my ESP32?

webhdx, 2020-09-11:
Hello
I'm reverse engineering a remote controller which connects to camera gimbal via BT. I want to simulate the gimbal using ESP32. They implemented some type of filtering so on the remote LCD you don't see incompatible devices, only particular gimbal model.

At first I thought it is using Manufacturer Data for that. Mobile app does that and by replicating Manufacturer Data the app sees my ESP32. It doesn't work with the remote tho.

I'm not sure if the remote works at this point... Is it possible to attach some kind of callback when the remote is scanning and asks my ESP32 for its info? This way I'd able to tell it's working and indeed scanning for devices.

Thanks in advance

chegewara, 2020-09-11:
Hi,
there is 2 ways ble can do scan:
- passive, ble device is just listening for packets in air,
- active, ble is listening packets and when find a packet then is sending request to that device for scan response packet

Honestly, i dont know about any option to have info when other device is requesting scan response (using active scan).

The thing is that your gimbal device can send 2 different advertising data, or maybe you didnt replicate scan response.

webhdx, 2020-09-11:
I only have screens from **nRF Connect** app that someone who has the gimbal sent me. I don't own it myself but I was curious to see if I'm able to sniff data from the remote controller. I think I replicated it well enough. The app checks Manufacturer Data, by setting it I'm able to make my ESP32 show up in the app scanning window. This made me think my remote is just broken and it's not receiving anything over BT. I think I will just buy the gimbal and do better research on how it introduces itself over BT.

chegewara, 2020-09-11:
Well, it may or not help you if you buy gimbal.
Lets consider situation, which is pretty common:
- controller is scanning for gimbal,
- found it, and is connecting,
- controller is sending very specific data and expecting very specific response.

In such situation you may try to use sniffer, but even then there is no guarantee you can reverse engineer it.
Or it is very simple option and you just have to replicate advertising packets, both.

webhdx, 2020-09-11:
Success. I was able to replicate the gimbal hardware using ESP32. First I used `hcidump` to see exact HCI packets which lead me to conclusion that the gimbal is advertising two packets. One with advertisement data and second with scan response. This indicates the remote controller is doing active scan so it was crucial to replicate proper Scan Response packet. It was possible with Arduino and BLE library.

closed 2020-09-11 by webhdx