nkolban/esp32-snippets #1014: BLE_Client unable to connect to ble_server if server is turned on after BLE_client

BabitaD, 2020-09-17:
Hello,

I am facing this issue of my esp32 not getting connected to server if I turn ON the server after more than 3 or 4 secs I turn ON esp32 client. 
I am trying to fetch data from a device which is continuously transmitting, The code works fine if I turn on the device before esp32 client and reconnects properly upon disconnection, It also works fine if I turn on the device within 3-4 seconds I turn on esp 32 client. If esp32 connects to the server once, It works perfectly fine thereafter i.e reconnects if the device is restarted
It is only when the device is turned ON a little more later than esp32 client that I don't get any output and it is stuck.

**These are few lines of verbose output when the code is working fine**
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Starting Arduino BLE Client application...
[V][BLEScan.cpp:195] start(): >> start(duration=5)
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: ScanEnd (0x3ffdd180), owner: <N/A> for start
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: ScanEnd (0x3ffdd180), owner: start
[V][BLEScan.cpp:227] start(): << start()
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 2 0x02
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: ScanEnd (0x3ffdd180), owner: start for start
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 7 0x07
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -61
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x6
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x06 (), length: 16, data: 6db8e67ea3934688074c355280cbeacd
[D][BLEAdvertisedDevice.cpp:447] setServiceUUID(): - addServiceUUID(): serviceUUID: cdeacb80-5235-4c07-8846-93a37ee6b86d
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 8, data: 748050338b180eca
[D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 748050338b180eca
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x9
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 7, data: 4d65646963616c
[D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: Medical
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x12
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x12 (), length: 4, data: 28002800
[D][BLEAdvertisedDevice.cpp:343] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xa
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x0a (), length: 1, data: 00
[D][BLEAdvertisedDevice.cpp:478] setTXPower(): - txPower: 0
BLE Advertised Device found: Name: Medical, Address: 50:33:8b:18:0e:ca, manufacturer data: 748050338b180eca, serviceUUID: cdeacb80-5235-4c07-8846-93a37ee6b86d, txPower: 0
[V][BLEScan.cpp:250] stop(): >> stop()
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: ScanEnd (0x3ffdd180), owner: start
[V][BLEScan.cpp:262] stop(): << stop()
[V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: ScanEnd (0x3ffdd180), owner: <N/A>
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEDevice.cpp:60] createClient(): >> createClient
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 18 0x12
[V][BLEDevice.cpp:66] createClient(): << createClient
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEClient.cpp:96] connect(): >> connect(50:33:8b:18:0e:ca)
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:593] addPeerDevice(): add conn_id: 0, GATT role: client
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegEvt (0x3ffdd630), owner: <N/A> for connect
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegEvt (0x3ffdd630), owner: connect
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: RegEvt (0x3ffdd630), owner: connect for connect
[... 41 lines cut ...]
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Starting Arduino BLE Client application...
[V][BLEScan.cpp:195] start(): >> start(duration=5)
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: ScanEnd (0x3ffdd184), owner: <N/A> for start
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: ScanEnd (0x3ffdd184), owner: start
[V][BLEScan.cpp:227] start(): << start()
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 2 0x02
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: ScanEnd (0x3ffdd184), owner: start for start
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 7 0x07
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***

Nothing gets printed after this
I tried updating the libraries, it did not work.
I am quite new to esp32 development, Am I something missing here?
Please Help.

This is my arduino code file.
[BLE_client_getData.txt](https://github.com/nkolban/esp32-snippets/files/5241763/BLE_client_getData.txt)

BabitaD, 2020-09-18:
Went through library codes, increased the value in this line of code
pBLEScan->start(5, false);
and got it working perfect. It was accepting connections till 5 sec due to this value.

closed 2020-09-18 by BabitaD

libasoles, 2020-10-16:
I tried like this and keeps on failing:

`advertiserScanner->start(60 * 5);
`

`advertiserScanner->start(60 * 5, false);
`

And actually, for me it fails just because of the mere presence of this line:
`BLEDevice::init("");`