nkolban/esp32-snippets #1015: BLE beacon tx power

MohammadSheayto, 2020-09-21:
Dear all,

I'm now working on advertising beacons to send some data on Arduino IDE.
I'm trying to increase the power using esp_err_t esp_ble_tx_power_set(esp_ble_power_type_t power_type, esp_power_level_t power_level) function on Arduino ID, but no change in the power, also as I see on the API reference that are used only after connection.
Is there any solution to increase the tx power without connection ?

chegewara, 2020-09-21:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/controller_vhci.html#_CPPv420ESP_BLE_PWR_TYPE_ADV

MohammadSheayto, 2020-09-23:
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/controller_vhci.html#_CPPv420ESP_BLE_PWR_TYPE_ADV

thank you!

closed 2020-09-23 by MohammadSheayto