nkolban/esp32-snippets #1016: esp_gatt_defs.h: No such file or directory

Eric97, 2020-10-06:
1. Created a separate project based on  [esp-idf-template](https://github.com/espressif/esp-idf-template) the cpp-utils with esp-idf framework. `hello-world` example compiled and executed without any problems. 

2. menuconfig enabled BLE stack.

__In file included from ../components/cpp_utils/BLEAdvertisedDevice.cpp:17:_
../components/cpp_utils/BLEAdvertisedDevice.h:12:10: fatal error: esp_gattc_api.h: No such file or directory
 #include <esp_gattc_api.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
[12/88] Building CXX object esp-idf/cpp_utils/CMake...es/__idf_cpp_utils.dir/BLECharacteristicMap.cpp.obj
FAILED: esp-idf/cpp_utils/CMakeFiles/__idf_cpp_utils.dir/BLECharacteristicMap.cpp.obj_

chegewara, 2020-10-07:
Did you enable BLE client in menuconfig?

Eric97, 2020-10-08:
Thanks. Managed to solve this problem. 

Should copy the entire repo of `esp32-snippets` instead of the [`cpp_utils`](https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils) only under the `esp-idf-templates/components`.

Think it is more clear to put this [readme](https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/README.md) at the root folder of the repo.

Eric97, 2020-10-08:
Sorry think I have to reopen this as the `cpp_utils` should be copied under the components. 
Yes, I have enabled the BLE. You may check sdkconfig as attached.

chegewara, 2020-10-09:
What IDF are you using, what build system (Cmake or make)?

Eric97, 2020-10-11:
I am using the latest `esp-idf` master. Ninja build system.

chegewara, 2020-10-11:
Great, now, do you have CmakeLists.txt in folder with this library?

Eric97, 2020-10-11:
Yes. Attached the CMakeLists.txt. Also the structure of my project:

```
├── build
│   ├── app-flash_args
│   ├── bootloader
│   ├── bootloader-flash_args
│   ├── bootloader-prefix
│   ├── build.ninja
│   ├── CMakeCache.txt
│   ├── CMakeFiles
│   ├── cmake_install.cmake
│   ├── compile_commands.json
│   ├── config
│   ├── config.env
│   ├── esp-idf
│   ├── flash_app_args
│   ├── flash_args
│   ├── flash_args.in
│   ├── flash_bootloader_args
│   ├── flasher_args.json
│   ├── flash_project_args
│   ├── kconfigs.in
│   ├── kconfigs_projbuild.in
│   ├── ldgen_libraries
│   ├── ldgen_libraries.in
│   ├── partition_table
│   ├── partition_table-flash_args
│   ├── project_description.json
│   ├── project_elf_src.c
│   ├── rules.ninja
│   └── x509_crt_bundle.S
├── CMakeLists.txt
├── components
│   └── cpp_utils
├── LICENSE
├── main
│   ├── CMakeLists.txt
│   ├── component.mk
│   ├── Kconfig.projbuild
│   └── main.c
├── Makefile
├── README.md
├── sdkconfig
└── sdkconfig.old
```
[CMakeLists.txt](https://github.com/nkolban/esp32-snippets/files/5361230/CMakeLists.txt)

chegewara, 2020-10-11:
You need another CmakeLists.txt in cpp_utils folder. Each component, in this case cpp_utils, requires to have its own CmakeLists.txt to get build.
It is not included in this library because it was prepared to build with old make.

Eric97, 2020-10-12:
Ah. thanks for the tip. I just did the normal `Make`, then seems all `bt` related stuff are built successfully. But came across another error: 
`components/cpp_utils/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared`

From this [issue933](https://github.com/nkolban/esp32-snippets/issues/933), need to use the `esp-idf` v3.3?

chegewara, 2020-10-12:
https://github.com/nkolban/esp32-snippets/issues/889

you will also have to fix one more change in esp-idf v4.x, but dont remember what. Its related to whitelist or something, adding one more parameter.

Also some related changes are here:
https://github.com/nkolban/esp32-snippets/pull/948/files